arduino/Arduino #6877: Arduino Time fails in October 31 (WeekDay)

aprada66, 2017-10-31:
Hi:

I'm programming a simple clock, using Arduino + LCD display + RTC chip.

I show the day of the week in the display, but actually the library fails in october 31th.

I try with october 30 and OK (display shows "monday"), november 1st and OK  (display shows "monday"), but october 31 shows "saturday"!!!.  I test 31/Oct/2016 and display shows "Friday"..... 31/Oct/2015 and fails too!

Please check, and correct if possible.  I hope updated library soon.  

Regards,

Albán

facchinm, 2017-10-31:
Hi @aprada66 ,
which library are you using exactly? Can you provide a link? If it's https://github.com/PaulStoffregen/Time you should report the bug there.
Thanks

aprada66, 2017-10-31:
Hi:

According Time.cpp file:

  **time.c - low level time and date functions
  Copyright (c) Michael Margolis 2009-2014**
...
...
...
  1.0  6  Jan 2010 - initial release
  1.1  12 Feb 2010 - fixed leap year calculation error
  1.2  1  Nov 2010 - fixed setTime bug (thanks to Korman for this)
  1.3  24 Mar 2012 - many edits by Paul Stoffregen: fixed timeStatus() to update
                     status, updated examples for Arduino 1.0, fixed ARM
                     compatibility issues, added TimeArduinoDue and TimeTeensy3
                     examples, add error checking and messages to RTC examples,
                     add examples to DS1307RTC library.
  1.4  5  Sep 2014 - compatibility with Arduino 1.5.7

facchinm, 2017-10-31:
It's definitely Paul's one https://github.com/PaulStoffregen/Time/blob/master/Time.cpp#L19 . 
Closing here, please reopen the issue on that repo, thanks!

closed 2017-10-31 by facchinm