arduino/Arduino #6878: Console does not use current font size

per1234, 2017-10-31:
Using Arduino IDE 1.9 Beta build 23 on Windows 7 64 bit

1. Change font size via keyboard shortcut, Edit menu, or File > Preferences > Editor font size
1. Sketch > Verify/Compile - the font size of the output in the console window uses the font size that was saved to preferences.txt the last time the Arduino IDE was closed rather than the current font size used in the editor.

per1234, 2017-11-02:
@facchinm just to note this isn't specific to the beta IDE. It also occurs with previous IDE versions (I just verified it occurs in 1.8.5 and 1.6.5). I noted 1.9 Beta build 23 to indicate that was the newest version I had confirmed the issue with but I should have also said it occurs with previous versions in case that caused confusion.

I find it useful to have a latest confirmed version noted when I go back to old open issues that I suspect were since solved because it allows me to quickly attempt to reproduce the issue with the same IDE version the user was using when they encountered the issue. If I can reproduce it with that version but not the latest I know the bug was fixed and the issue can be closed.

facchinm, 2017-11-02:
Yep, got it, I only added the tag as a placeholder to notify (myself) if something is still broken on the beta IDE :wink: