arduino/Arduino #6885: Math macros in Arduino.h

iacopobaroncini, 2017-11-03:
Hi folks,

I know it has been discussed already here and there...

I find semantically wrong the fact that the following math functions are implemented as macros:
**abs**, **min**, **max**, **constrain**, **round**.

I would say the same about **bitWrite** if a look at its macro definition.

It is well known that using these function the way the rest of the world does, introduces faults, and therefore failures.

Writing something like this (example) **abs(cos(x) - 123 * sq(sin(x))** would just introduce **redundant calculations**, a crazy delay in this case...
Writing something like this (example) **abs(i++)** just introduce a **fault**... it does something different.

Assuming that using **template** and **inline functions** would solve the issue with no extra executable space taken, I would like to know if there is any other good reason why those dangerous 
definitions are still in force.

Kind regards,
Iacopo.

PaulStoffregen, 2017-11-05:
I can confirm this multiple expression evaluation problem has come up for people using Teensy.

The solution is not so simple, because the C++ std namespace also has min & max, and bringing in those headers causes all sorts of other compatibility headaches.  These functions/macros are also needed for plain C, which is seldom used in the Arduino world, but failure to consider it will break some programs.

Here's a conversation from earlier this year about some of the details:

https://forum.pjrc.com/threads/44596-Teensyduino-1-37-Beta-2-(Arduino-1-8-3-support)?p=145150&viewfull=1#post145150

iacopobaroncini, 2017-11-05:
All crear Paul.
Many thanks.
I actually had problems (with abs and round in first instance) while working with Arduino boards (micro and mega).

cousteaulecommandant, 2017-12-25:
Maybe make both the templated function and a macro that calls the templated function? That would avoid the multiple call issue AND the namespace clash.
```c++
template<typename T> T &Max(T &a, T &b) { return (a>=b) ? a : b; }
template<typename T> const T &Max(const T &a, const T &b) { return (a>=b) ? a : b; }
#define max(a, b) Max((a), (b))
```
Some related discussion (about `round()` specifically): #6098