arduino/Arduino #6894: Add Library request: MFUthings

beetlikeyg087, 2017-11-06:
Hello,

Please add this library to Library Manager

https://github.com/beetlikeyg087/MFUthings

thank you.

per1234, 2017-11-06:
You need to create a release or tag that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

beetlikeyg087, 2017-11-07:
I have updated information,

can you please check again?

best regards.

cmaglie, 2017-11-09:
Done!

closed 2017-11-09 by cmaglie

njh, 2020-04-17:
Hi @beetlikeyg087

It looks like your library has now been deleted?
https://github.com/beetlikeyg087/MFUthings

Any background as to why?

nick.