arduino/Arduino #6896: Add library request: KeyboardAzertyFr

martin-leo, 2017-11-07:
Hello,

I made a Keyboard library fork to support the french azerty keyboard layout.

You can check it at this address : https://github.com/martin-leo/KeyboardAzertyFr

Best regards.

per1234, 2017-11-07:
Please update the `name` and version values in your library.properties file and then create a release or tag that matches the `version` value. Since there is already a tag for the current `version` value (which was inherited from the fork when the library was still named "Keyboard") you should increment the version before creating the new release.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

martin-leo, 2017-11-07:
I'm not sure of exactly what to do but I tried to follow your instructions. Is this Okay ?  : https://github.com/martin-leo/KeyboardAzertyFr/releases/tag/1.0.1-fr

per1234, 2017-11-07:
Here's my concern: In [Semver ](http://semver.org/)when you append a hypen and then a string to a version number that's considered a pre-release. So that means that 1.0.1 actually has precedence over 1.0.1-fr. This could possibly cause two different problems:

The permanent name chosen by Library Manager for the library.
From https://github.com/arduino/Arduino/issues/5058#issuecomment-227785970
>Previously it was possible to change name between releases, now this is no more allowed and libraries must keep the name they used when registered. If the name in library.properties is changed the indexer will override this setting and always use the name used during registration.
...
 I've selected the name used in the latest release, in bold in the table above, as the official name.

In the [GitHub releases page](https://github.com/martin-leo/KeyboardAzertyFr/releases) 1.0.1-fr is indeed shown as the latest release but I don't know whether the Library Manager indexer uses Semver version order to determine what is considered the "latest release".

The other concern would be that, even if the correct name is chosen, when someone installs a library via Library Manager the highest version number is chosen by default, which should be 1.0.1, not 1.0.1-fr. If you want to install a different version then you need to choose it from the version menu and even then if it's not the latest version and the user has update notifications enabled then they will get nagged by the IDE to "update" the library even though in this case that would mean a downgrade since it would mean replacing 1.0.1-fr with 1.0.1.

That's all just my suspicions. I don't know whether any of that would happen or whether Library Manager will be able to ignore the inherited releases. @cmaglie is the expert on this and he can give a definitive verdict on this when he gets a chance to review this request.

If it does end up being a problem I see a couple of options to fix it:
1. Delete the tags inherited from the Keyboard library.
2. Change your release version to be indisputably higher than the previous tag. If you still want to preserve the Keyboard library version number you could add that as a pre-release (even though that's not really the intended usage). For example: 1.0.2-keyboard.1.0.1 or using the build metadata scheme: 1.0.2+keyboard.1.0.1

martin-leo, 2017-11-07:
Thank you for this long (not too mutch), documented and useful answer.

Based on this information I chose to :
1. delete the inherited tags and releases (they are release sof the original and I think it makes sense to delete them here)
2. make a new release with 1.0.1 as a version number to remain coherent with the original library (as it is simply a kind of localization).

Do you think this will do ? I do not mind to do some additional modification if it can be improved.

per1234, 2017-11-07:
You forgot to change the `version` value in library.properties before you created the 1.0.1 release. Other than that I think any potential problems have been avoided by removing the inherited tags.

Please comment once you have resolved the `version` value issue.

martin-leo, 2017-11-07:
I changed the ```version``` value in library.properties, then deleted the release then recreated it.

cmaglie, 2017-11-09:
Done!

closed 2017-11-09 by cmaglie