arduino/Arduino #6898: RTC_DS3231 does not name a file type

jsteele09, 2017-11-07:
Previously working sketches are no longer compiling. Even the supplied example sketches no longer work giving the same error. This is on Arduino editor online and the local Arduino 1.8.3. See attached for example code I was referring to.
[DS3231_dateformat.txt](https://github.com/arduino/Arduino/files/1450792/DS3231_dateformat.txt)

facchinm, 2017-11-10:
Hi @jsteele09 , which version of the IDE was compiling this sketch successfully? The error means that "RTClib.h" cannot be found by the builder; I'm closing the issue here, you should reopen it on https://github.com/arduino/Arduino-builder with the following information:
- IDE version in which this sketch was compiling
- **Verbose** output of the compilation (enable it on file/preferences)
- Check if it works with Beta IDE (https://www.arduino.cc/en/Main/Software#beta)

Thanks!

closed 2017-11-10 by facchinm

jsteele09, 2017-11-10:
The original IDE version that I successfully compiled was 1.8.3. Using the web version on create.arduino.cc I was able to get my sketches to compile once more. All I had to do to accomplish this was change RTC_DS3231 rtc; to DS3231rtc; Everything works and it keeps time as it should. I am not sure if a library changed automatically online, but I don't see why mine quit on the desktop unless I may have updated that library myself without realizing it.

facchinm, 2017-11-10:
Yep, probably the local library you have installed contains some modifications over the regular one (which is "shipped" by Create). You can try moving it outside your sketchbook and redownload it via Library Manager to check if they're different.

Giovani-Delgado, 2018-11-10:
Please help me!
I have not been able to compile with the web editor for a week.
It bears the following message:
No matching function for call to'DS3231 (const uint8_t &, const uint8_t &)'
I already uninstalled the DS3231 library and reinstalled. Unsuccessfully.
The usage examples for the DS3231 also carry the same message...
In IDE 1.8.5 it compiles perfectly.
Thank you very much for your attention!

per1234, 2018-11-10:
@Giovani-Delgado please post about this to the Create > Editor section of the Arduino Forum:
http://forum.arduino.cc/index.php?board=101.0

That is the best place for us to provide support for your problem. It is also the place to report bugs in the Arduino Web Editor in case we determine that is the problem.

Make sure to provide all the information we will need to reproduce the problem, including the complete error output. This will allow us to help you get your problem solved quickly.