arduino/Arduino #6902: Library examples page implies the Wire library is not compatible with Due

per1234, 2017-11-08:
From the Wire Library section of https://www.arduino.cc/en/Tutorial/LibraryExamples:
>For all Arduino boards, BUT Arduino DUE.

To me, that reads as saying the Wire library is not compatible with Due but I'm pretty sure that's wrong because Arduino SAM Boards does contain a Wire library, the example sketches are functionally identical to those in the examples page, and the Wire library reference page does mention the Due.

Originally reported at http://forum.arduino.cc/index.php?topic=510365

SimonePDA, 2018-03-06:
As documented in the Wire Library page, it IS compatible with the Arduino DUE as well. Corrected the wrong statement in the Library Examples page.

Thanks

per1234, 2019-03-26:
Re-opening because the erroneous statement remains:
[image]

SimonePDA, 2019-03-26:
For some reason the page was "recovered" from an earlier edit and the correcion got lost. Redone.

closed 2019-03-26 by SimonePDA