arduino/Arduino #6903: Arduino 1.8.6 Error Downloading 3rd party boards

NathanielJS1541, 2017-11-08:
When trying to download 3rd party boards, I put the link (for example I will use https://adafruit.github.io/arduino-board-index/package_adafruit_index.json) into the Additional board URLs in the preferences, and then go to the board manager and I get this error:

Error downloading https://adafruit.github.io/arduino-board-index/package_adafruit_index.json.

Arduino boards are installed just fine, and Arduino is whitelisted in my security software (Kaspersky). I have also tried disabling my security software completely.

I have completely uninstalled Java and Arduino, and re-installed them and I still get the same error. I have trawled through so many forums but none of the methods posted there seem to work.

I am using Windows 10 Home x64 bit Version 10.0.15063 Build 15063.

bluemurder, 2017-11-08:
Same error is showed when internet connection is not available or there's a proxy server blocking some network requests. Check if you are in one of above conditions

NathanielJS1541, 2017-11-08:
I definitely have an internet connection, and I'm pretty sure there isn't a proxy server as I can download and install the arduino boards that are already listed in the boards manager. I can access all of the links without an issue in my browser too.

facchinm, 2017-11-10:
Hi @NathanielJS1541, would you mind testing the Beta version of the IDE (https://www.arduino.cc/en/Main/Software#beta) ? They can happily coexist side by side so it should be easy to test it. 
Anyway, it looks like a firewall/proxy issue. 
/self-reminder We should add some more information to the error output to ease troubleshooting :slightly_smiling_face:

NathanielJS1541, 2017-11-10:
I tried "arduino-PR-beta1.9-BUILD-24" but unfortunately got the same result. The thing that I'm confused about is that I can install any of the boards that are listed in the boards manager by default, so the IDE can access the internet, but for some reason the 3rd party boards never install...

facchinm, 2017-11-10:
DNS problems on .io domains in your world region maybe?

NathanielJS1541, 2017-11-10:
Could be... I did read a forum where Japan I think it was had that issue. Although I can access the .json from my browser? Not sure if this is relevant to that point though as I don't know enough in that area. I am in the UK and haven't found anyone experiencing the same issue but I'm not ruling it out.

feikname, 2017-11-10:
>  I did read a forum where Japan I think it was had that issue

I believe you're referring to #6767?

NathanielJS1541, 2017-11-10:
Yes that's the one. Unfortunately it looks less similar than I remember it being, I may try using a VPN to the US to see if anything changes.

NathanielJS1541, 2017-11-13:
I just gave up in the end and "refreshed" my windows install, basically just uninstalled everything and reinstalled windows. Everything works now, so must have been an error in my registry or settings...

closed 2017-11-13 by NathanielJS1541

BILLGLENN, 2018-09-10:
The Windows reinstall will work for a while. I have been battling this issue for 3 weeks now. At one point, I installed my cloned C:/ drive that I had created 2 weeks before this problem began. After swapping out the problematic C:/ with the clone, everything worked just great. So, I thought whatever was the problem was now banished from my system. Not so fast! About 1 week later, all of a sudden, the EXACT problem was back:

Error downloading https://adafruit.github.io/arduino-board-index/package_adafruit_index.json
Error downloading https://www.adafruit.com/package_adafruit_index.json

Keep in mind that NOTHING about the internet network or connectivity has been modified. Everyday, just routine usage of the Arduino UI and the same Adafruit board project was conducted. Then, one day, one just wakes up, turns on the computer, loads Arduino and BAM! --- the Adafruit boards will not load and thus there is no access to them. It's a complete disruption of what has been a long and prosperous development cycle. But no more. I am literally at my wits end trying to guess why this boards manager cannot get the Adafruit boards. They are the only ones that are blocked for some unknown reason. And, just for those who may be thinking viruses/malware etc., a full scan has been accomplished and there is nothing there.  I hope there is some IT techy who will figure this one out. (It occurs across platforms) and it's only the Adafruit, third party files that are inaccessible. Thanks in advance.

BILLGLENN, 2018-09-10:
After a couple hours (at least), I solved this bad boy. Having a known-good clone of my C:/ drive from 3 weeks ago (before this issue reared its ugly head), I was able to go through the ARDUINO15 directory of files and do a substitution, 1-by-1, of the files there. The annoying thing about other methods I tried was that once fixed, the problem would re-surface because one can edit the ARDUINO15 directory files only when the UI is not running. Else, the application will write over the "fix" and put you back to square one. Anyway, no soap opera. The bad files were the package_adafruit_index.json.sig.tmp and package_index.json.sig.   Instead of deleting them, I just renamed them ....sig.xxx  and .tmp.xxx  followed by restarting the Arduino UI.  Presto... no more issues downloading the Adafruit boards from GitHub.  Damn, I am really glad to solve this one. It seemed to come back to life after updating the boards. 

Enjoy. 
PS:  To confirm, I re-instated the bad files and, presto, the problem was back. Then it took all of 60 seconds to fix a problem that has cost me many, many days of lost time.  I wish I knew more details but I'm not wasting any more time going through each of those files to figure out why they are so troublesome.  Cao

ChrisBall, 2019-04-09:
Just a quick message to confirm I was facing this issue and BILLGLENN's solution worked for me. For anyone else wondering, the ARDUINO15 directory he refers to is in c:\users\your username\appdata\local\arduino15. I just removed all files ending ".tmp".

BILLGLENN, 2019-04-09:
Chris,
Thanks for the confirmation & letting me know it 
worked for you.
Cheers,
Bill

  ----- Original Message ----- 
  From: ChrisBall
  To: arduino/Arduino
  Cc: BILLGLENN ; Comment
  Sent: Tuesday, April 09, 2019 7:09 AM
  Subject: Re: [arduino/Arduino] Arduino 1.8.6 
Error Downloading 3rd party boards (#6903)

  Just a quick message to confirm I was facing 
this issue and BILLGLENN's solution worked for me. 
For anyone else wondering, the ARDUINO15 directory 
he refers to is in c:\users\your 
username\appdata\local\arduino15. I just removed 
all files ending ".tmp".

CrystalSunrise, 2019-07-02:
This just saved my life. Thank you!

BILLGLENN, 2019-07-02:
Thanks for letting me know you had success.

Regards,

Bill
  ----- Original Message ----- 
  From: CrystalSunrise
  To: arduino/Arduino
  Cc: BILLGLENN ; Comment
  Sent: Tuesday, July 02, 2019 2:07 PM
  Subject: Re: [arduino/Arduino] Arduino 1.8.6 
Error Downloading 3rd party boards (#6903)

  This just saved my life. Thank you!

Jerar, 2019-07-10:
Thanks for this thread, renaming *.tmp files in Ardiono15 folder solves the problem with downloading adafruit boards

jimfrankfort, 2024-05-15:
Thanks for the above.  I was getting the same errors described with the adafruit...json file.  When I looked at the preferences.txt file in ...AppData\Local\Arduino15\preferences.txt, I noticed that there was a space in the URL for the index.....even though it was not in what I pasted into the text box for additional board manager URLs.  I corrected it in the .txt file.  Result, no more error downloading the index.json and the adafruit boards were there.

rrwakc, 2024-06-16:
I have manualy saved json in C:\Users\User Nmae\AppData\Local\Arduino15

and then followed this guide(it is for Heltec boards but would assume it is similar for others), to install boards manualy using git.
https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series/blob/master/InstallGuide/windows.md