arduino/Arduino #6904: Rename existing library in Library manager

Marzogh, 2017-11-08:
Hi guys, I’m looking at possibly renaming my SPIFlash library to something else as it appears to cause issues for users using @LowPowerLab ‘s SPIFlash library. (Felix's library has been around for a lot longer than mine)

Would it be possible re-name my library and not break the upgrade path (through library manager) for current users?

cmaglie, 2017-11-09:
> Would it be possible re-name my library and not break the upgrade path (through library manager) for current users?

No, existing users will need to reinstall the library with the new name.

Moreover, you must ask here to change the name in the library manager, you can't do it autonomously.

Marzogh, 2017-11-09:
Thanks for the information. I'll see/do what needs to be done. :)

closed 2017-11-09 by Marzogh