arduino/Arduino #6907: Arduino IDE 1.8.[45] will not open on Mac OSX -- similar to #5109

unixwizard107, 2017-11-09:
Ok this has been happening since I switched systems.  I have upgraded Java, and IDE.
I have:  rm ~/Library/Ardui*/preferences.txt  ~/Library/Preferences/cc.arduino.Arduino.*

Clearly, some file is corrupted, but I have no idea which one, since it does not say what file its trying to read.

With 1.8.4 I get:  
% /Applications/Arduino.app/Contents/MacOS/Arduino
java.io.CharConversionException: Invalid UTF-32 character 0x102c003e(above 10ffff)  at char #1559, byte #6239)
	at com.fasterxml.jackson.core.io.UTF32Reader.reportInvalid(UTF32Reader.java:189)
	at com.fasterxml.jackson.core.io.UTF32Reader.read(UTF32Reader.java:150)
	at com.fasterxml.jackson.core.json.ReaderBasedJsonParser.loadMore(ReaderBasedJsonParser.java:153)
	at com.fasterxml.jackson.core.json.ReaderBasedJsonParser._skipWSOrEnd(ReaderBasedJsonParser.java:2017)
	at com.fasterxml.jackson.core.json.ReaderBasedJsonParser.nextToken(ReaderBasedJsonParser.java:577)
	at com.fasterxml.jackson.databind.ObjectMapper._initForReading(ObjectMapper.java:3776)
	at com.fasterxml.jackson.databind.ObjectMapper._readMapAndClose(ObjectMapper.java:3721)
	at com.fasterxml.jackson.databind.ObjectMapper.readValue(ObjectMapper.java:2796)
	at cc.arduino.contributions.packages.ContributionsIndexer.parseIndex(ContributionsIndexer.java:204)
	at cc.arduino.contributions.packages.ContributionsIndexer.mergeContributions(ContributionsIndexer.java:143)
	at cc.arduino.contributions.packages.ContributionsIndexer.parseIndex(ContributionsIndexer.java:108)
	at processing.app.BaseNoGui.initPackages(BaseNoGui.java:487)
	at processing.app.Base.<init>(Base.java:257)
	at processing.app.Base.main(Base.java:140)

With 1.8.5 I gets:
% /Applications/Arduino.app/Contents/MacOS/Arduino
java.io.CharConversionException: Invalid UTF-32 character 0x102c003e(above 10ffff)  at char #1559, byte #6239)
	at com.fasterxml.jackson.core.io.UTF32Reader.reportInvalid(UTF32Reader.java:189)
	at com.fasterxml.jackson.core.io.UTF32Reader.read(UTF32Reader.java:150)
	at com.fasterxml.jackson.core.json.ReaderBasedJsonParser.loadMore(ReaderBasedJsonParser.java:153)
	at com.fasterxml.jackson.core.json.ReaderBasedJsonParser._skipWSOrEnd(ReaderBasedJsonParser.java:2017)
	at com.fasterxml.jackson.core.json.ReaderBasedJsonParser.nextToken(ReaderBasedJsonParser.java:577)
	at com.fasterxml.jackson.databind.ObjectMapper._initForReading(ObjectMapper.java:3776)
	at com.fasterxml.jackson.databind.ObjectMapper._readMapAndClose(ObjectMapper.java:3721)
	at com.fasterxml.jackson.databind.ObjectMapper.readValue(ObjectMapper.java:2796)
	at cc.arduino.contributions.packages.ContributionsIndexer.parseIndex(ContributionsIndexer.java:204)
	at cc.arduino.contributions.packages.ContributionsIndexer.mergeContributions(ContributionsIndexer.java:143)
	at cc.arduino.contributions.packages.ContributionsIndexer.parseIndex(ContributionsIndexer.java:108)
	at processing.app.BaseNoGui.initPackages(BaseNoGui.java:487)
	at processing.app.Base.<init>(Base.java:261)
	at processing.app.Base.main(Base.java:144)

facchinm, 2017-11-10:
Hi @unixwizard107 ,the offending file is surely a contribution json which resides in `~/Library/Arduino15` folder. Delete all jsons in there (and signatures files and incomplete downloads) and the IDE should start again correctly. Let me know if it works!

unixwizard107, 2017-11-10:
Thank you
rm  package_esp8266com_index.json* 
seems to have been the solution

facchinm, 2017-11-10:
Sure, you're right about the error message which could be much more informative. If you want to propose a patch we'll be glad to merge it!

closed 2017-11-10 by facchinm