arduino/Arduino #6914: Mention processor ressources in reference documentation (like millis() command etc.)

dl1dmw, 2017-11-10:
Neebies and hobyists like me like to know the internal processor ressources used by a command like millis() or other c commands. In case of millis() it should be mentioned the timer used in the reference doc. In other cammands it would be helpful to know the estimated amount of space.
Also in Arduino libs: They should mention used internal processor ressources to avoid conflicts.

kengdahl, 2018-03-06:
Hey. Here is a article about timers you might find interesting: 
[https://www.arduino.cc/en/Tutorial/SecretsOfArduinoPWM](https://www.arduino.cc/en/Tutorial/SecretsOfArduinoPWM)

SimonePDA, 2018-03-06:
The documentation is made to be easy to read and understand; the more tech-heavy topics are dealt on our forum and on many dedicated pages around the net. You may find the details you are looking for just visiting our forum and by googling your specific topic.

Thanks for your request.

closed 2018-03-06 by SimonePDA