arduino/Arduino #6917: Arduino 1.7.10 NO Borad manager

elwolv, 2017-11-10:
IDE 1.7.10   tools/board  
No Board manager appears at top of the drop menu?
how can we add the board manager?

install IDE 1.8 version has lot of errors on Java modules?

per1234, 2017-11-10:
You're correct that Arduino IDE 1.7.10 does not have Boards Manager. There's absolutely nothing we can do about that.

What we can do is try to fix any bugs that you might be having with the latest version of  the Arduino IDE but we can't do that unless you provide detailed bug reports.

The first step is for you to confirm the issues still occur with the **beta** version of the Arduino IDE, which you can download for free from https://www.arduino.cc/en/Main/Software#beta.

After doing so you should read the relevant parts of https://github.com/arduino/Arduino/blob/master/CONTRIBUTING.md and follow those guidelines in submitting bug reports.

Please note that this issue tracker is only to be used for reporting bugs in the Arduino IDE. If you're having problems with your own code, including code that compiled in 1.7.10 but does not with the latest Arduino IDE version, then you should instead post to the Arduino Forum after reading this:
https://forum.arduino.cc/index.php?topic=148850.0

closed 2017-11-10 by per1234

elwolv, 2017-11-12:
Arduino 1.8 version downloaded as zip file and opened correctly, but not as download-run will not work.

problem solved using the zip downloaded file...