arduino/Arduino #6919: Errors on Product offering page

artisticforge, 2017-11-12:
Hello

https://www.arduino.cc/en/Main/Products needs updating.

At least three errors have been found.
https://www.arduino.cc/en/Main/ArduinoBoard101 is now 404.
The Arduino Board 101 has also been removed from the Arduino Store.
https://store.arduino.cc/homepage/arduino-mkrfox1200 is now 404.
The MKR Fox 1200 has also been removed from the Arduino Store.
https://www.arduino.cc/en/Main/ArduinoGemma
The Gemma is flagged as 'Retired"
The Gemma should be moved to the Retired section.

per1234, 2017-11-12:
When I set my location to Europe / Africa the pages https://www.arduino.cc/en/Main/ArduinoBoard101 and https://store.arduino.cc/homepage/arduino-mkrfox1200 load fine. When I set my location to America / Asia / Oceania I get a 404 or am redirected to the store home page (depending on whether I change location with the page already loaded or load the URL after changing location).

artisticforge, 2017-11-15:
the Yun Shield  has also disappeared from the USA Arduino Store.
The link https://www.arduino.cc/en/Main/ArduinoYunShield is now https://store.arduino.cc/usa/no-route

artisticforge, 2017-11-15:
in the link https://store.arduino.cc/usa/arduino-usb-host-shield the images of the board have been Photoshop to remove "arduino.org" from the images. This is arduino.org behavior and should not be arduino.cc behavior. Please arduino.cc do not start deceiving us or lying to us.

per1234, 2019-03-26:
Reopening because https://www.arduino.cc/en/Main/ArduinoYunShield still goes to https://store.arduino.cc/usa/no-route when your location is set to America/Asia/Oceania.

The other links are working fine for either location setting now.

per1234, 2020-09-29:
I notice the "ARDUINO YÚN SHIELD" bug was fixed, but now there are new problems:

---
MKR FOX 1200 (back again): https://store.arduino.cc/homepage/arduino-mkrfox1200

should be: https://store.arduino.cc/arduino-mkr-fox-1200-1408

---
Arduino Fundamentals bundle: https://store.arduino.cc/arduino-fundamentals-bundle

should be:  https://store.arduino.cc/fundamentals-bundle

---
Arduino M0 Pro: https://www.arduino.cc/en/Main/ArduinoBoardM0PRO

should be: https://store.arduino.cc/m0-pro

---
Arduino Fio: https://www.arduino.cc/en/Main/ArduinoBoardFio

should be (only works when location set to "America, Asia, Oceania", so that's another bug): https://store.arduino.cc/usa/arduino-fio

---

Arduino USB2Serial Converter (only when location set to "America, Asia, Oceania"): https://www.arduino.cc/en/Main/MiniUSB

product link that only works when location set to "Europe and Africa": https://store.arduino.cc/usb-serial-converter

per1234, 2020-10-05:
Everything in my previous comment has now been fixed except for:

>( only works when location set to "America, Asia, Oceania", so that's another bug): https://store.arduino.cc/usa/arduino-fio

Since the Fio is a fairly obscure retired product, I'm going to go ahead and call this fixed!

closed 2020-10-05 by per1234