arduino/Arduino #6921: Data types in reference section; suggestion to add size_t

sterretje, 2017-11-12:
As title says

Add a data type size_t to the data type section; maybe add a hint about sizeof to determine what the size of size_t is so people can determine for themselves.

per1234, 2017-11-12:
Hi @sterretje thanks for your suggestions. In the future please submit issues related to the Language Reference pages to the [arduino/reference-en](https://github.com/arduino/reference-en) repository. Issues regarding non-Language Reference documentation pages should still be reported in this repository.

I've moved this to https://github.com/arduino/reference-en/issues/228. Any input from others regarding this proposal would be welcome on that thread.

closed 2017-11-12 by per1234

sterretje, 2017-11-13:
It's getting confusing ;-)