arduino/Arduino #6926: MacOS High Sierra Compile Issue

jjoker0110, 2017-11-14:
Hey all, 
just updated my mac to High Sierra 10.13.1 (17B48) this morning and after updating nothing would compile, not even the example scripts. Here is the error I got when trying to verify code:

`failed MSpanList_Insert 0x308000 0x166be44f5b32 0x0
fatal error: MSpanList_Insert

runtime stack:
runtime.throw(0x2a950b)
	/usr/local/go/src/runtime/panic.go:491 +0xad fp=0x7ffeefbff2f0 sp=0x7ffeefbff2c0
runtime.MSpanList_Insert(0x2c7028, 0x308000)
	/usr/local/go/src/runtime/mheap.c:692 +0x8f fp=0x7ffeefbff318 sp=0x7ffeefbff2f0
MHeap_FreeSpanLocked(0x2c3c20, 0x308000, 0x100)
	/usr/local/go/src/runtime/mheap.c:583 +0x163 fp=0x7ffeefbff358 sp=0x7ffeefbff318
MHeap_Grow(0x2c3c20, 0x8, 0x0)
	/usr/local/go/src/runtime/mheap.c:420 +0x1a8 fp=0x7ffeefbff398 sp=0x7ffeefbff358
MHeap_AllocSpanLocked(0x2c3c20, 0x1, 0x0)
	/usr/local/go/src/runtime/mheap.c:298 +0x365 fp=0x7ffeefbff3d8 sp=0x7ffeefbff398
mheap_alloc(0x2c3c20, 0x1, 0x12, 0x0)
	/usr/local/go/src/runtime/mheap.c:190 +0x121 fp=0x7ffeefbff400 sp=0x7ffeefbff3d8
runtime.MHeap_Alloc(0x2c3c20, 0x1, 0x10000000012, 0x10219)
	/usr/local/go/src/runtime/mheap.c:240 +0x66 fp=0x7ffeefbff438 sp=0x7ffeefbff400
MCentral_Grow(0x2cb998, 0x0)
	/usr/local/go/src/runtime/mcentral.c:197 +0x8b fp=0x7ffeefbff4a0 sp=0x7ffeefbff438
runtime.MCentral_CacheSpan(0x2cb998, 0x0)
	/usr/local/go/src/runtime/mcentral.c:85 +0x167 fp=0x7ffeefbff4d8 sp=0x7ffeefbff4a0
runtime.MCache_Refill(0x304000, 0x12, 0x0)
	/usr/local/go/src/runtime/mcache.c:90 +0xa0 fp=0x7ffeefbff500 sp=0x7ffeefbff4d8
runtime.mcacheRefill_m()
	/usr/local/go/src/runtime/malloc.c:368 +0x57 fp=0x7ffeefbff520 sp=0x7ffeefbff500
runtime.onM(0x221810)
	/usr/local/go/src/runtime/asm_amd64.s:273 +0x9a fp=0x7ffeefbff528 sp=0x7ffeefbff520
runtime.mallocgc(0x120, 0x1bd640, 0x0, 0x0)
	/usr/local/go/src/runtime/malloc.go:178 +0x849 fp=0x7ffeefbff5d8 sp=0x7ffeefbff528
runtime.newobject(0x1bd640, 0x304000)
	/usr/local/go/src/runtime/malloc.go:353 +0x49 fp=0x7ffeefbff600 sp=0x7ffeefbff5d8
runtime.newG(0x27daa)
	/usr/local/go/src/runtime/proc.go:233 +0x2a fp=0x7ffeefbff618 sp=0x7ffeefbff600
allocg(0x2b65e0)
	/usr/local/go/src/runtime/proc.c:925 +0x1f fp=0x7ffeefbff628 sp=0x7ffeefbff618
runtime.malg(0x8000, 0x2b6680)
	/usr/local/go/src/runtime/proc.c:2106 +0x1f fp=0x7ffeefbff658 sp=0x7ffeefbff628
runtime.mpreinit(0x2b6a40)
	/usr/local/go/src/runtime/os_darwin.c:137 +0x27 fp=0x7ffeefbff670 sp=0x7ffeefbff658
mcommoninit(0x2b6a40)
	/usr/local/go/src/runtime/proc.c:201 +0xc9 fp=0x7ffeefbff698 sp=0x7ffeefbff670
runtime.schedinit()
	/usr/local/go/src/runtime/proc.c:138 +0x55 fp=0x7ffeefbff6c0 sp=0x7ffeefbff698
runtime.rt0_go(0x7ffeefbff6f0, 0x18, 0x7ffeefbff6f0, 0x0, 0x18, 0x7ffeefbff8f8, 0x7ffeefbff930, 0x7ffeefbff93c, 0x7ffeefbff94c, 0x7ffeefbff956, ...)
	/usr/local/go/src/runtime/asm_amd64.s:95 +0x116 fp=0x7ffeefbff6c8 sp=0x7ffeefbff6c0
arduino-builder returned 2

Error compiling.
`

This isn't anything to do with the type of board I'm using because it fails before I even get to any part where I upload code to an arduino.

Let me know if you need any more information on this bug or if I did something incorrectly.

per1234, 2017-11-14:
Which version of the Arduino IDE are you using?

jjoker0110, 2017-11-14:
Arduino 1.6.7

jjoker0110, 2017-11-14:
Upgraded to 1.8.5 and it works now, thought I had the latest version already but didn't. Closing issue

closed 2017-11-14 by jjoker0110

per1234, 2017-11-14:
Duplicate of https://github.com/arduino/Arduino/issues/5360