arduino/Arduino #6929: Incorrect behavior of Arduino icon pinned to Windows taskbar

vzhukovskyy, 2017-11-14:
When I pin the Arduino icon to taskbar and click it, it launches the IDE but also creates duplicate icon on the taskbar.

Steps: 
1. Go to C:\Program Files (x86)\Arduino, right-click on arduino.exe and select 'Pin to taskbar'
2. Click on the taskbar icon, wait for the IDE to appear
3. Notice the duplicate icon on the taskbar. If right-click it, the context menu has only one choice - close the window.

Expected: 
Launching the IDE should reuse the icon and not create another one.

Analysis and solution:
It looks like arduino.exe is just a launcher for java application. I tried to create an application shortcut to javaw.exe passing all the parameters found for Arduino's javaw in the sysinternal's Process Explorer - but I could not because of 260-symbols limitation. So I tried to reduce the command length by using wildcards in classpath instead of listing jars one by one, but the length was still more then 260. Then I extracted a part of command line parameters to a system variable ARDUINO_PARAMS and created application shortcut to 
```"C:\Program Files (x86)\Arduino\java\bin\javaw.exe" -splash:"C:\Program Files (x86)\Arduino/lib/splash.png" %ARDUINO_PARAMS% processing.app.Base```. Pinned it to taskbar and it worked as expected, no additional icons were created. Using wildcards in the classpath was still a required step because environment variable has also 260 symbols limitation.

This solution is rather a workaround, maybe there is a better approach. But I personally would not mind if Arduino installer created ARDUINO_PARAMS in my environment variables for the sake of nicely behaving taskbar icon.

Sod-Almighty, 2019-02-22:
It's been over a year. Isn't it time you fixed this?

tiboine, 2019-11-05:
two years now ... 
and we still can't pin it to the taskbar with right-clicking the app-icon.

stoneRdev, 2021-09-28:
4 years later, still not fixed? 

It's laughable that they labelled this issue as a "feature request" when it should be a "bug" (in my opinion, and an easy one to fix at that). 
The fact that no one from the team has added any updates or referenced this at all lets me know that this isn't important to them. Looks like its up to the community to do something. maybe I can write a script that can be ran post-install to enable it to be pinnable.

Sod-Almighty, 2021-09-28:
I agree it is a bug. And, to my mind, it's one caused by their insistence on writing the application in Java. If they used a "real" language - that is, one that designed for writing Windows applications and which uses standard Windows controls, such as C++ or C# - then this problem would not arise.

Java was, is, and always will be horrible. Programs written in Java never look and feel like native applications; and you get these strange "edge cases" - like the taskbar pinning problem - where Java and Windows simply refuse to talk to each other sensibly.

JMK97, 2021-11-01:
try to get all rights to your microsoft apps like described in the link below. Then go to ArduinoLLC.../arduino.exe and pin it to your taskbar

get all rights: 

https://praxistipps.chip.de/windows-10-wo-ist-der-speicherort-von-apps_42720

douwe2, 2023-03-10:
Still waiting for that fix 6 years later. I wonder how many people times have been wasted.
90% Fix:
If you only want to have the icon changed do the following:
Install Resource Hacker and change the icons from java.exe, javaw.exe and javaws.exe to the arduino icon.
These files are typicall found in: C:\Program Files (x86)\Arduino\java\bin
If you still see the old java icon you must restart the computer to have it take effect (windows huh)

If you also want to pin arduino to the taskbar you also need to do the following:
Create an arduino.bat file in your Documents folder (for example) with the follwoing content:
"C:\Program Files (x86)\Arduino\arduino.exe"
Next  create a shortcut in "C:\Users\eman\AppData\Roaming\Microsoft\Windows\Start Menu\Programs"
with as 'Target' C:\Windows\explorer.exe C:\Users\eman\Documents\arduino.bat
Now go the windows 'Start' menu and you can now select 'Pin to taskbar'.
Shortcoming is that after starting Arduino this way you will se two Arduino icons on the taskbar.
If that bothers you then you could right click on the taskbar icon that start arduin, then right click once more on arduino and then change the icon.

per1234, 2023-03-11:
Closing as fixed in Arduino IDE 2.x. If anyone is affected by this bug, please update your Arduino IDE to the latest version. The download links are listed here:

https://www.arduino.cc/en/software

closed 2023-03-11 by per1234