arduino/Arduino #6931: CLI Suggestion : Export compiled binary

dimitre, 2017-11-15:
It would be great to have a parameter that compiles and export compiled binary only.

Example of usefulness: inside Arduino IDE the max baud rate I can upload a sketch to ESP8266 is 115200 and using esptool.py I can upload using baud 3000000.
Thank you

matthijskooijman, 2017-11-15:
There is an "Export Compiled Binary" in the "Sketch" menu. Is that what you mean?

dimitre, 2017-11-15:
Sorry @matthijskooijman it is about CLI, command line arduino options. I'll edit my post for clarification

dimitre, 2018-07-28:
Just as an usage example, one line to compile and export binary from a sketch
```
/Applications/Arduino.app/Contents/MacOS/Arduino --export ArduinoSketch.bin --board esp8266:esp8266:generic:CpuFrequency=80,FlashFreq=40,FlashMode=dio,UploadSpeed=230400,FlashSize=512K64,ResetMethod=ck,Debug=Disabled,DebugLevel=None____ ArduinoSketch.ino
```

andrethomas, 2018-08-30:
Did you get this to work?

dimitre, 2018-08-30:
@andrethomas no. in fact I was meaning a new parameter in Arduino CLI to be able to export a binary compiled file via command line. This is an improvement suggestion.

andrethomas, 2018-08-30:
Oh ok, I already do that with platformio but wanted to do it with arduino - somehow the linking scripts work different for the esp

andrethomas, 2018-08-30:
~/arduino/arduino -v --board esp8266:esp8266:d1_mini:CpuFrequency=80,FlashSize=4M1M --verify sketch.ino --preserve-temp-files --pref build.path=/build/

Causes the .bin file to be in the /build/ folder specified so there's a way to do it I guess

pkoevesdi, 2020-05-25:
Thank You for the solution.
Let me add this, since I stumbled upon it:
- You can't set the build path to the sketch path
https://github.com/arduino/arduino-builder/issues/4#issuecomment-140402264
- To get the `--board` parameter run
`arduino --verify your_sketch.ino -v`
and fish out the value behind `-fqbn=`

matthijskooijman, 2020-05-25:
Thanks for the suggestion.

Since this issue was created, https://github.com/arduino/arduino-cli was created, which will be the backend for the java IDE as well in the future. Since that offers a much more complete CLI interface, I expect that the java IDE commandline interface will become deprecated at some point, and will likely not be developed any further. For that reason, I'm going to close this as wontfix.

As for this particular request, I suspect `arduino-cli` does not implement it yet, but the same approach as above would work: Run a compile with a fixed build path, and extract the binary from the buildpath afterwards. See also https://arduino.github.io/arduino-cli/latest/commands/arduino-cli_compile/

closed 2020-05-25 by matthijskooijman

cmaglie, 2020-05-25:
Actually it's the opposite: `arduino-cli` will always export the compiled binary inside the `sketch` folder.

To be precise until `arduino-cli` 0.10.0 the binary was exported in `sketch/sketch.ino.board-id.hex` where "board-id" is the board identifier.
We recently merged a patch that changes this behaviour and now the cli export in `sketch/build/board-id/sketch.ino.hex` that is muuuch more clean, this will be available when `arduino-cli` >=0.11.0 (not yet but soon to be released).

pkoevesdi, 2020-05-25:
Thanks for pointing out.
Is andrethomas' (and my current) solution using the java IDE commandline interface? (I was not well aware of what's happening when calling arduino from command line.)

HARIS4820, 2022-07-27:
i need help to export compiled binary for DOIT ESP32 DEVKIT v1 board. I dont know how to select this board. help me Thanks in Advance