arduino/Arduino #6932: Can't compile from 1.8.5 on Windows 10 Fall Update when sketchbook located on OneDrive

ngmacha, 2017-11-16:
I'm frequently using different computers for Arduino development, so I keep a sketchbook on my OneDrive, and it worked without problem till latest Windows 10 Fall Update. After the this Windows version update (I'm currently on build 16299.rs3_release.170928-1534), even when I set the whole sketchbook folder as "Always available offline", Arduino 1.8.5 can't compile any sketch located on OneDrive, or that uses any library located under the sketchbook. If I move the sketch/library to local hard-drive then everything compiles OK. Browsing/opening/editing/saving on OneDrive has no issues.
Looks like this happens only with Arduino 1.8 (tried 1.8.3 and 1.8.5), but Arduino 1.6.5 on the same machine doesn't have any issues and compiles everything without problems.
So what 1.8 have different from 1.6 that prevent it from compiling sketches/libraries on OneDrive? Is ther a workaround for this, or we'll have to wait till next release?

sterretje, 2017-11-16:
Did you see http://forum.arduino.cc/index.php?topic=509222.0 ? It might be the solution for your issue.

ngmacha, 2017-11-16:
Thanks for pointing to that thread, @sterretje! However there is no solution yet... Turning off the OneDrive's "Files on Demain" is not a solution, as older Arduino versions still works with the option on. Subscribed to #6867 though.

ngmacha, 2017-11-16:
Hm, just tried Arduino 1.6.12, and it behaves exactly like 1.8.5!! So there must be something fundamentally changed in Arduino IDE between 1.6.5 and 1.6.12 that causing this...

ngmacha, 2017-11-16:
Just downloaded Arduino 1.6.6 to test, cannot compile the OneDrive located sketch! So 1.6.6 is where all started!! Maybe not a coincidence, but it is the first version that the startup splash shows "Arduino + Genuino". Reading the Release Note also shows several thing firstly introduced in this version, among others these two might be the culprit:
- Switched to Java 8
- Introducing arduino-builder

I'm not a developer, but hopefully these finding may help to have a fix quicker...

matthijskooijman, 2017-11-16:
Both of these changes sound like they could be related.

When you say "cannot compile", what do you mean exactly? Any error messages? Perhaps you can enable verbose compilation and paste the entire output here?

ngmacha, 2017-11-16:
The error is simple - it just can't see the file, either ino sketch or the library:

> Arduino: 1.8.5 (Windows 10), Board: "Arduino/Genuino Uno"
>
> C:\Apps\arduino-latest\arduino-builder -dump-prefs -logger=machine -hardware C:\Apps\arduino-latest\hardware -hardware C:\Users\HaMac\AppData\Local\Arduino15\packages -hardware C:\Users\HaMac\OneDrive\Documents\Ardunio\hardware -tools C:\Apps\arduino-latest\tools-builder -tools C:\Apps\arduino-latest\hardware\tools\avr -tools C:\Users\HaMac\AppData\Local\Arduino15\packages -built-in-libraries C:\Apps\arduino-latest\libraries -libraries C:\Users\HaMac\OneDrive\Documents\Ardunio\libraries -fqbn=arduino:avr:uno -ide-version=10805 -build-path C:\Users\HaMac\AppData\Local\Temp\arduino_build_721501 -warnings=default -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=C:\Users\HaMac\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.1.1 -prefs=runtime.tools.avrdude.path=C:\Users\HaMac\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino9 -prefs=runtime.tools.avr-gcc.path=C:\Users\HaMac\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2 -verbose C:\Users\HaMac\OneDrive\Documents\Ardunio\BlinkTwoLEDs\BlinkTwoLEDs.ino
>
> readlink C:\Users\HaMac\OneDrive\Documents\Ardunio\BlinkTwoLEDs\BlinkTwoLEDs.ino: The system cannot find the file specified.
>
> Error compiling for board Arduino/Genuino Uno.

matthijskooijman, 2017-11-16:
Seems definitely related to the arduino-builder introduction. I've opened https://github.com/arduino/arduino-builder/issues/254 for this, so I'm closing this issue. Thanks for the info so far, let's continue discussing there.

closed 2017-11-16 by matthijskooijman