arduino/Arduino #6934: Digital Read Serial typo

pleabargain, 2017-11-16:
URL
https://www.arduino.cc/en/Tutorial/DigitalReadSerial

bord 
should be 
board

Once the bord has read the input, make it print this information back to the computer as a decimal value. You can do this with the command Serial.println() in our last line of code:
should be
Once the board has read the input, make it print this information back to the computer as a decimal value. You can do this with the command Serial.println() in our last line of code:

kengdahl, 2018-02-19:
Fixed

per1234, 2019-03-26:
Re-opening because the typo is still there:
[image]

closed 2020-05-23 by per1234