arduino/Arduino #6951: Search function drops first character into code window

fredlcore, 2017-11-20:
When using the keyboard shortcut CMD+F on my Mac to quickly access the search window and then quickly continuing to enter the search text, the first character is still entered in the code window, not in the search box. I'd say that one has to wait roughly half a second until it is safe to continue typing. 

This is especially problematic as the search box may hide the part of the code window where the cursor is located. In case I had highlighted some part of code before, it will get replaced with the first character of the search text and I may find out only later when I complile the code next time. By then, I may still undo until the part in question is restored, but this will also undo all my other changes since then. 

System: MacOS High Sierra
Arduino IDE: 1.8.5

facchinm, 2017-11-20:
Hi @fredlcore , would you mind testing Beta 1.9.0 and report if the behaviour is different (hopefully better) ? Thanks!

fredlcore, 2017-11-20:
Hi @facchinm , thanks for the prompt response - yes, the behaviour is still the same :(...

per1234, 2017-12-15:
Arduino IDE 1.9.0 Beta build 26 on Windows 7 64 bit

I don't know whether I'm encountering the same bug. For me with the beta build the Find window never gets focus no matter how long I wait. I don't have this problem with Arduino IDE 1.8.5 or Arduino IDE 1.8.6 Hourly Build 2017/11/13 11:33, the cursor is immediately moved to the Find input field with those IDE versions.

facchinm, 2017-12-15:
Hi @per1234 ,
it's surely related with a couple of commits: https://github.com/arduino/Arduino/commit/65103aa7c6cf24beba06ce95d34dcf5b78206e49 and https://github.com/arduino/Arduino/commit/0d50f0bb6c6baf8bf750cd268451418c8cd81e21, both targeting another issue I can't find right now (probably because it had the wrong description).
If you could provide a small screencast of what is happening it would be very cool.

facchinm, 2017-12-15:
Got it: https://github.com/arduino/Arduino/pull/6603

per1234, 2017-12-15:
The requested screencast. This is using Ctrl+F but the same happens with Edit > Find (both of which work as expected with 1.8.5 and the hourly build)
[image]

fredlcore, 2017-12-15:
Just to mention that I assume that this is most likely not related to the bug I repoted above because my observation is not limited to any current version but has been around on my Mac since Arduino 1.8.5 (and probably before).

facchinm, 2017-12-15:
@per1234 on Linux it grabs the focus as soon as you open it and loses when the editor is clicked (as expected). So the problem is probably generated by https://github.com/arduino/Arduino/commit/0d50f0bb6c6baf8bf750cd268451418c8cd81e21 which doesn't take into account the "just spawned" event. I'll take a look on a Win machine to make sure I understood the problem correctly.

per1234, 2017-12-15:
OK, sorry if I got this issue off topic. I seemed to remember this had been reported previously but this was the only one I found other than https://github.com/arduino/Arduino/issues/6408, which is different. I'm happy to move this to a new issue on request.

I should have also mentioned that I also encountered my issue on Arduino IDE 1.9.0 beta build 26 before I updated to build 30 to verify the issue still occurred.