arduino/Arduino #6953: Forces the computer to restart

iclojano15, 2017-11-21:
This takes 5 seconds to make the computer to go crazy and says it has to force restart because of a problem 
I checked the holes and everything but it won't stop its crashing can someone help me?

per1234, 2017-11-21:
Please provide more details:
- Is the 5 seconds relative to starting the Arduino IDE or plugging the USB cable on your Arduino board in to the computer?
- Which version of the Arduino IDE are you using?
- Which operating system are you using?

per1234, 2018-04-28:
@iclojano15 we're still waiting for more details from you regarding this issue. Are you still having this problem?

per1234, 2018-07-26:
Closing due to lack of response. @iclojano15 if you can provide the requested information I'll be happy to reopen this.

closed 2018-07-26 by per1234