arduino/Arduino #6960: Library examples don't show after installing via Add .ZIP

per1234, 2017-11-23:
Using Arduino IDE 1.9,0 Beta build 25 with Windows 7 64 bit

1. Download https://github.com/arduino-libraries/UnoWiFi-Developer-Edition-Lib/archive/master.zip (or any other library that was not previously installed and has examples)
1. Sketch > Include Library > Add .ZIP Library... > select the downloaded file > Open
1. File > Examples - the expected "Arduino Uno WiFi Dev Ed Library" menu item does not appear
1. Restart the Arduino IDE
1. File > Examples - the expected "Arduino Uno WiFi Dev Ed Library" menu item does appear

This issue does not occur with Arduino IDE 1.8.6 Hourly Build 2017/11/13 11:33 or Arduino IDE 1.8.5.

This issue does not occur with the Beta build when installing libraries via Library Manager.

closed 2018-05-09 by facchinm

rbpasker, 2022-10-15:
This seems to be broken (again) in my version. I installed a zip library and it didnt show up until I restarted the IDE.

Version: 2.0.0
Date: 2022-09-14T07:08:28.861Z (4 weeks ago)
CLI Version: 0.27.1 [a900cfb2]

per1234, 2022-10-15:
Hi @rbpasker. The bug in Arduino IDE 2.0.0 is tracked here:

https://github.com/arduino/arduino-ide/issues/659