arduino/Arduino #6961: [Library Manager] Please add Libraries from stm32duino

cparata, 2017-11-23:
Hi,
Please add to Library Manager (sensors components, connectivity,...)

https://github.com/stm32duino/Proximity_Gesture
https://github.com/stm32duino/VL6180X
https://github.com/stm32duino/X-NUCLEO-IKS01A1
https://github.com/stm32duino/X-NUCLEO-IKS01A2
https://github.com/stm32duino/X-NUCLEO-6180XA1
https://github.com/stm32duino/X-NUCLEO-53L0A1
https://github.com/stm32duino/X-NUCLEO-NFC01A1
https://github.com/stm32duino/X-NUCLEO-NFC04A1
https://github.com/stm32duino/X-NUCLEO-IDB05A1
https://github.com/stm32duino/X-NUCLEO-LED61A1
https://github.com/stm32duino/X-NUCLEO-IKA01A1
https://github.com/stm32duino/X-NUCLEO-IHM02A1

All have a release.

Thanks in advance

fpistm, 2017-11-24:
Could you add also this one:
https://github.com/stm32duino/WiFi-ISM43362-M3G-L44
Thanks in advance

cmaglie, 2017-11-27:
Done!

closed 2017-11-27 by cmaglie