arduino/Arduino #6965: ctags pattern is missing on openSUSE 42.3

AEtherC0r3, 2017-11-25:
I've installed the Arduino IDE using the guide https://playground.arduino.cc/Linux/OpenSUSE
But, when I'm trying to compile a sketch e.g. the Blink example, it throws the following error
```
ctags pattern is missing

Error compiling for board Arduino/Genuino Uno.
```

The binaries from https://www.arduino.cc/en/Main/Software work as expected

PaulStoffregen, 2017-11-25:
You need to report this problem to OpenSUSE.

Closing this issue would be appropriate, since it has nothing to do with upstream Arduino which you've already confirmed works.

closed 2017-11-27 by facchinm

thild, 2020-04-28:
> You need to report this problem to OpenSUSE.
> 
> Closing this issue would be appropriate, since it has nothing to do with upstream Arduino which you've already confirmed works.

See https://build.opensuse.org/package/show/CrossToolchain:avr/Arduino#comment-1131218

thild, 2020-04-28:
> I've installed the Arduino IDE using the guide https://playground.arduino.cc/Linux/OpenSUSE
> But, when I'm trying to compile a sketch e.g. the Blink example, it throws the following error
>
> ```
> ctags [...]

See https://build.opensuse.org/package/show/CrossToolchain:avr/Arduino#comment-1131218