arduino/Arduino #6968: [Library Manager] Non interrupting lighting library for single LEDS

SethSenpai, 2017-11-27:
[https://github.com/SethSenpai/singleLEDLibrary](https://github.com/SethSenpai/singleLEDLibrary)

Please add this library. A non interrupting library for single LED lighting behaviour.

per1234, 2017-11-27:
You need to create a release or tag that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

SethSenpai, 2017-11-27:
Whoops my bad. 
[https://github.com/SethSenpai/singleLEDLibrary/releases/tag/v1.0](https://github.com/SethSenpai/singleLEDLibrary/releases/tag/v1.0)

cmaglie, 2017-11-27:
Done!

closed 2017-11-27 by cmaglie