arduino/Arduino #6970: MD_MAX72xx library in Library Manager

MajicDesigns, 2017-11-27:
I have had to rebuild the repository at [](https://github.com/MajicDesigns/MD_MAX72XX) and it was missing/offline for a while. Is there anything that needs to be relinked/renewed from your side so that the new repo works properly with the IDE LIbrary Manager?

per1234, 2017-11-28:
I do notice that only the 2.6.0 version is in the Library Manager index.

Unrelated to that issue, you updated your `version` value in library.properties after the 2.10.0 release:
https://github.com/MajicDesigns/MD_MAX72XX/blob/v2.10.0/library.properties#L2
which will cause the 2.10.0 release to not get added to the library manager index.

MajicDesigns, 2017-11-28:
OK. I will correct the library.properties on the week end (travelling at the moment). IS this the only fix required for this?

cmaglie, 2017-12-04:
Hi @MajicDesigns, sorry for the delay...

> Is there anything that needs to be relinked/renewed from your side so that the new repo works properly with the IDE LIbrary Manager?

No, the repository is already indexed but the library is not published because it contains executables inside:

```
JOB 792 - 2017/12/04 23:26:04 Scraping https://github.com/MajicDesigns/MD_MAX72XX.git
JOB 792 - 2017/12/04 23:26:05 Checking out tag: v2.10.0
JOB 792 - 2017/12/04 23:26:05 Error syncing library: *.exe files found, skipping
JOB 792 - 2017/12/04 23:26:05 Checking out tag: v2.2
JOB 792 - 2017/12/04 23:26:05 Error syncing library: Error generating library from repo: can't read library.properties: open /home/jenkins/jenkins/jobs/Libraries/workspace/gitclones/github.com/MajicDesigns/MD_MAX72XX/library.properties: no such file or directory
JOB 792 - 2017/12/04 23:26:05 Checking out tag: v2.3
JOB 792 - 2017/12/04 23:26:05 Error syncing library: Error generating library from repo: can't read library.properties: open /home/jenkins/jenkins/jobs/Libraries/workspace/gitclones/github.com/MajicDesigns/MD_MAX72XX/library.properties: no such file or directory
JOB 792 - 2017/12/04 23:26:05 Checking out tag: v2.4
JOB 792 - 2017/12/04 23:26:05 Error syncing library: Error generating library from repo: can't read library.properties: open /home/jenkins/jenkins/jobs/Libraries/workspace/gitclones/github.com/MajicDesigns/MD_MAX72XX/library.properties: no such file or directory
JOB 792 - 2017/12/04 23:26:05 Checking out tag: v2.5
JOB 792 - 2017/12/04 23:26:05 Error syncing library: Error generating library from repo: can't read library.properties: open /home/jenkins/jenkins/jobs/Libraries/workspace/gitclones/github.com/MajicDesigns/MD_MAX72XX/library.properties: no such file or directory
JOB 792 - 2017/12/04 23:26:05 Checking out tag: v2.7
JOB 792 - 2017/12/04 23:26:05 Release MD_MAX72XX:2.6.0 already loaded, skipping
JOB 792 - 2017/12/04 23:26:05 Checking out tag: v2.8
JOB 792 - 2017/12/04 23:26:05 Release MD_MAX72XX:2.6.0 already loaded, skipping
JOB 792 - 2017/12/04 23:26:05 Checking out tag: v2.9
JOB 792 - 2017/12/04 23:26:05 Error syncing library: *.exe files found, skipping
JOB 792 - 2017/12/04 23:26:05 Checking out tag: v2.9.1
JOB 792 - 2017/12/04 23:26:05 Error syncing library: *.exe files found, skipping
```

we do not allow that as a security policy.
By removing any `.exe` file (and by retagging the revision) the library should be published automatically at the next indexing round that happens every hour.

closed 2017-12-04 by cmaglie