arduino/Arduino #6971: Library Manager: please add ESP32 HTTP Update

suculent, 2017-11-27:
https://github.com/suculent/esp32-http-update

Thanks.

per1234, 2017-11-28:
The `version` value in your library.properties file must match the release/tag version but yours does not:
https://github.com/suculent/esp32-http-update/blob/2.1.144/library.properties#L2
Please update the `version` value and then either redo the release or create a new release.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

suculent, 2017-11-28:
Thanks, fixed.

per1234, 2017-11-28:
You change the version after the release so the version value in the release is still wrong: 
https://github.com/suculent/esp32-http-update/blob/2.1.144/library.properties#L2
You need to either redo the release or update the version again and then create a new release that matches the version.

cmaglie, 2017-12-04:
Done!

closed 2017-12-04 by cmaglie