arduino/Arduino #6973: java.lang.NullPointerException	at cc.arduino.packages.uploaders.SerialUploader.uploadUsingPreferences(SerialUploader.java:79)

zoubata, 2017-11-29:
hi, 
I devellop a 3th party board.
A follow the instruction of https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5-3rd-party-Hardware-specification
I have compare to MattairTech_Arduino release but I have still this error : 

"C:\Users\....\AppData\Local\Temp\arduino_build_684435/Blink.ino.elf"
   text	   data	    bss	    dec	    hex	filename

   4292	     32	    228	   4552	   11c8	C:\Users\pvalleau\AppData\Local\Temp\arduino_build_684435/Blink.ino.elf

Sketch uses 4292 bytes (1%) of program storage space. Maximum is 253952 bytes.
java.lang.NullPointerException
		at cc.arduino.packages.uploaders.SerialUploader.uploadUsingPreferences(SerialUploader.java:79)
	at cc.arduino.UploaderUtils.upload(UploaderUtils.java:78)
	at processing.app.SketchController.upload(SketchController.java:713)
	at processing.app.SketchController.exportApplet(SketchController.java:686)
	at processing.app.Editor$DefaultExportHandler.run(Editor.java:2149)
	at java.lang.Thread.run(Thread.java:745)

it look like that the plateform isn't loaded, why ?
how can i start arduino tool to be verbose to identify where the issue is ?

facchinm, 2017-12-01:
Hi @zoubata , how are you specifying the uploader?
If you are using one of our loaders (`bossac` for example) or MattairTech's one?
On our `boards.txt` you should specify the provider of the tool, too (see https://github.com/mattairtech/ArduinoCore-samd/blob/master/boards.txt#L629 and https://github.com/mattairtech/ArduinoCore-samd/blob/master/boards.txt#L734).
Of course, if you depend on another core's tools, the other core should be installed.
Let me know if my guess is right :wink:

zoubata, 2017-12-01:
Hi,
I try to fit with mattairtech_xeno : 

zoubworld_Pilo.upload.protocol=sam-ba
zoubworld_Pilo.bootloader.tool=arduino:openocd
...
zoubworld_Pilo.menu.bootloader.8kb.upload.tool=Zoubworld_Arduino:bossac
....
and I have on :  C:\Users\....\AppData\Local\Arduino15\packages\zoubworld_Arduino\tools\bossac\1.7.0-zoubworld-1\bossac.exe

json file are like this : 
 "tools": [
      {
        "name": "bossac",
        "version": "1.7.0-zoubworld-1",
        "systems": [
...
and the plateform file have this :  

#
# BOSSA
#
tools.bossac.path={runtime.tools.bossac-1.7.0-zoubworld-1.path}
tools.bossac.cmd=bossac
tools.bossac.cmd.windows=bossac.exe

tools.bossac.upload.params.verbose=-i -d
tools.bossac.upload.params.quiet=
tools.bossac.upload.pattern="{path}/{cmd}" {upload.verbose} --port={serial.port.file} -U {upload.native_usb} -i -e -w -v "{build.path}/{build.project_name}.bin" -R

tools.bossac_remote.upload.pattern=/usr/bin/run-bossac {upload.verbose} --port=ttyATH0 -U {upload.native_usb} -e -w -v /tmp/sketch.bin -R

tools.bossac.network_cmd={runtime.tools.arduinoOTA.path}/bin/arduinoOTA
tools.bossac.upload.network_pattern="{network_cmd}" -address {serial.port} -port 65280 -username arduino -password "{network.password}" -sketch "{build.path}/{build.project_name}.bin" -upload /sketch -b

Regarding the line of the exception( SerialUploader.java:79)
" prefs.putAll(targetPlatform.getTool(tool));" (based on git master)
It looks like that the targetPlatform isn't loaded from BaseNoGui.getTargetPlatform();

Any other advice ?
Regards

facchinm, 2017-12-01:
If you want to share your json I'll test it locally; in the meantime, you can check if the target package loaded correctly by launching `arduino --get-pref` from commandline (`arduino_debug.exe` if you are on Windows).
If there is no `runtime.tools.bossac-1.7.0-zoubworld-1.path` entry it means that something in the json is not ok. Are you loading it locally (via `file://` url or via network)?

zoubata, 2017-12-08:
So i finished the release system and the server : 
you can install from : http://lfbn-1-14132-125.w90-12.abo.wanadoo.fr/~zoubworld_Arduino/files/Release/package_Zoubworld_index.json
to install : board managers : zoub... SAM M0+ ..... Pilo
to compil & download : 
The config is
board Pilo
Microcontroller SAMC21J18
USB config : USB disable
Clock : Internal oscillator
Serial config : One usart one wire one SPI
programmer ArduinoISP

per1234, 2017-12-08:
Does you mean you solved the problem @zoubata?

zoubata, 2017-12-08:
no the problem is stil here.
but I can share this you the project, you can try.

per1234, 2017-12-08:
I did this:
1. Add http://lfbn-1-14132-125.w90-12.abo.wanadoo.fr/~zoubworld_Arduino/files/Release/package_Zoubworld_index.json to **File > Preferences > Additional Boards Manager URLs**.
2. Use Boards Manager to install zoubworld SAM M0+ Boards 1.6.17.
3. Tools > Board > zoubworld Pilo
4. File > New
5. Sketch > Verify/Compile

I get this:
```
C:\Users\per\AppData\Local\Arduino15\packages\zoubworld_Arduino\hardware\samd\1.6.17\variants\Pilo\variant.cpp:156:1: error: 'TC0' was not declared in this scope

 TC0, TC1, TCC2, TC3, TC4, 
 ^

C:\Users\per\AppData\Local\Arduino15\packages\zoubworld_Arduino\hardware\samd\1.6.17\variants\Pilo\variant.cpp:156:6: error: 'TC1' was not declared in this scope

 TC0, TC1, TCC2, TC3, TC4, 
      ^
```

zoubata, 2017-12-08:
you have to select this : 
-  board Pilo
- Microcontroller SAMC21J18 becarefull SAM   "C" 21J18 
- USB config : USB disable
- Clock : Internal oscillator
- Serial config : One usart one wire one SPI
- programmer ArduinoISP

sorry,
today i support only the SAMC21J18

facchinm, 2017-12-11:
It looks like http://lfbn-1-14132-125.w90-12.abo.wanadoo.fr/~zoubworld_Arduino/files/Release/package_Zoubworld_index.json is not accessible, would you mind sharing it in another way (as a github release, maybe)? Thanks

zoubata, 2017-12-11:
server is up again
you can try

facchinm, 2017-12-11:
Ok, got the error.
You don't need to specify the provider if you are using your own tool; replacing `zoubworld_Arduino:bossac` with `bossac` in `boards.txt` will point to the correct folder (and solves the NPE).

closed 2017-12-11 by facchinm

zoubata, 2017-12-11:
ok thank's,
It work now 👍 
see the log : 
"C:\Users\pvalleau\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1/bin/arm-none-eabi-size" -B "C:\Users\pvalleau\AppData\Local\Temp\arduino_build_966381/Blink.ino.elf"
   text	   data	    bss	    dec	    hex	filename

   4404	     32	    732	   5168	   1430	C:\Users\pvalleau\AppData\Local\Temp\arduino_build_966381/Blink.ino.elf

Sketch uses 4404 bytes (1%) of program storage space. Maximum is 253952 bytes.
Forcing reset using 1200bps open/close on port COM14
PORTS {COM1, COM3, COM14, } / {COM1, COM3, COM14, } => {}
PORTS {COM1, COM3, COM14, } / {COM1, COM3, COM14, } => {}
PORTS {COM1, COM3, COM14, } / {COM1, COM3, COM14, } => {}
PORTS {COM1, COM3, COM14, } / {COM1, COM3, COM14, } => {}
PORTS {COM1, COM3, COM14, } / {COM1, COM3, COM14, } => {}
PORTS {COM1, COM3, COM14, } / {COM1, COM3, COM14, } => {}
PORTS {COM1, COM3, COM14, } / {COM1, COM3, COM14, } => {}
PORTS {COM1, COM3, COM14, } / {COM1, COM3, COM14, } => {}
PORTS {COM1, COM3, COM14, } / {COM1, COM3, COM14, } => {}
PORTS {COM1, COM3, COM14, } / {COM1, COM3, COM14, } => {}
PORTS {COM1, COM3, COM14, } / {COM1, COM3, COM14, } => {}
PORTS {COM1, COM3, COM14, } / {COM1, COM3, COM14, } => {}
PORTS {COM1, COM3, COM14, } / {COM1, COM3, COM14, } => {}
PORTS {COM1, COM3, COM14, } / {COM1, COM3, COM14, } => {}
PORTS {COM1, COM3, COM14, } / {COM1, COM3, COM14, } => {}
PORTS {COM1, COM3, COM14, } / {COM1, COM3, COM14, } => {}
PORTS {COM1, COM3, COM14, } / {COM1, COM3, COM14, } => {}
PORTS {COM1, COM3, COM14, } / {COM1, COM3, COM14, } => {}
PORTS {COM1, COM3, COM14, } / {COM1, COM3, COM14, } => {}
PORTS {COM1, COM3, COM14, } / {COM1, COM3, COM14, } => {}
Uploading using selected port: COM14
C:\Users\pvalleau\AppData\Local\Arduino15\packages\zoubworld_Arduino\tools\bossac\1.7.0-zoubworld-1/bossac.exe -i -d --port=COM14 -U true -i -e -w -v C:\Users\pvalleau\AppData\Local\Temp\arduino_build_966381/Blink.ino.bin -R 
Set binary mode
Send auto-baud
Set binary mode
readWord(addr=0)=0x20007ffc
readWord(addr=0xe000ed00)=0x410cc601
readWord(addr=0x41002018)=0x11010200
version()=v2.0 [Arduino:XYZ] Nov 27 2017 15:39:07
chipId=0x11010000
Connected at 115200 baud
readWord(addr=0)=0x20007ffc
readWord(addr=0xe000ed00)=0x410cc601
readWord(addr=0x41002018)=0x11010200
Atmel SMART device 0x11010000 found
write(addr=0x20004000,size=0x34)
writeWord(addr=0x20004030,value=0x10)
writeWord(addr=0x20004020,value=0x20008000)
Device       : ATSAMC21J18A
readWord(addr=0)=0x20007ffc
readWord(addr=0xe000ed00)=0x410cc601
readWord(addr=0x41002018)=0x11010200
Chip ID      : 11010000
version()=v2.0 [Arduino:XYZ] Nov 27 2017 15:39:07
Version      : v2.0 [Arduino:XYZ] Nov 27 2017 15:39:07
Address      : 8192
Pages        : 3968
Page Size    : 64 bytes
Total Size   : 248KB
Planes       : 1
Lock Regions : 16
Locked       : readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
none
readWord(addr=0x41004018)=0
Security     : false
Boot Flash   : true
readWord(addr=0x40000834)=0
BOD          : false
readWord(addr=0x40000834)=0
BOR          : false
Arduino      : FAST_CHIP_ERASE
Arduino      : FAST_MULTI_PAGE_WRITE
Arduino      : CAN_CHECKSUM_MEMORY_BUFFER
Erase flash
chipErase(addr=0x2000)
done in 1.030 seconds

Write 4436 bytes to flash (70 pages)
write(addr=0x20004400,size=0x1000)
writeBuffer(scr_addr=0x20004400, dst_addr=0x2000, size=0x1000)

[===========================   ] 91% (64/70 pages)write(addr=0x20004400,size=0x180)
writeBuffer(scr_addr=0x20004400, dst_addr=0x3000, size=0x180)

[==============================] 100% (70/70 pages)
done in 0.700 seconds

Verify 4436 bytes of flash with checksum.
checksumBuffer(start_addr=0x2000, size=0x1000) = 57b6
checksumBuffer(start_addr=0x3000, size=0x154) = fc74
Verify successful
done in 0.019 seconds
CPU reset.
readWord(addr=0)=0x20007ffc
readWord(addr=0xe000ed00)=0x410cc601
readWord(addr=0x41002018)=0x11010200
writeWord(addr=0xe000ed0c,value=0x5fa0004)