arduino/Arduino #6978: Dead link on lilypad simple page

sterretje, 2017-11-30:
I hope this is the correct section.

On the "lilypad simple" product page https://store.arduino.cc/lilypad-arduino-simple is a link that refers to http://lilypadarduino.org/?p=149 (first link under "need help?") which is 'dead'.

per1234, 2017-11-30:
Here's the lilypadarduino.org page on the Wayback Machine:
http://web.archive.org/web/20161018192320/http://lilypadarduino.org/?p=149
There is also an arduino.cc page for the LilyPad boards, including the LilyPad Simple: https://www.arduino.cc/en/Guide/ArduinoLilyPad
but it has different content than the lilypadarduino.org page.

q2dg, 2017-12-12:
Nowadays best information source for Lilypads is Sparkfun: https://www.sparkfun.com/lilypad_sewable_electronics

kengdahl, 2018-01-02:
I have removed the link until other content is made to replace it.

closed 2018-01-02 by kengdahl