arduino/Arduino #6979: SPI library seems to be outdated in Arduino

quirxi, 2017-11-30:
Operating System:
Linux debian 4.9.0-3-amd64 #1 SMP Debian 4.9.30-2+deb9u5 (2017-09-19) x86_64 GNU/Linux
debian version: 9.2 (stretch)

Arduino Version: 2:1.0.5+dfsg2-4.1

Problem:
Trying to compile a sketch that uses the [Adafruit-MAX31855-library](https://github.com/adafruit/Adafruit-MAX31855-library) and included "SPI.h" gave an error message saying that the symbol **"beginTransaction"** in  **"SPI.h"** could not be found. I googled around and found out that other people had this problem, too. So I installed the newer version of the [SPI](https://github.com/arduino/Arduino/tree/master/hardware/arduino/avr/libraries/SPI) library and everything compiled and worked flawlessly. 
Also in the source code of the newer version of SPI.h it says that the old methods like "setBitOrder" or "setDataMode" are deprecated and the newer method "beginTransaction" should be used instead.

So I wonder if the SPI library that gets delivered with the arduino ide is a bit outdated and if yes if it could be replaced by the newer version ?

per1234, 2017-11-30:
When you use apt-get to install the Arduino IDE you get a very outdated and possibly non-standard version of the Arduino IDE, thus the problem you encountered. Please always download the Arduino IDE from https://www.arduino.cc/en/Main/Software to get the latest official release.

closed 2017-11-30 by per1234

quirxi, 2017-11-30:
Ok, thanks !