arduino/Arduino #6981: CRC doesn't match on Intel Curie (Arduino 101)

frankracis, 2017-12-02:
Trying to install the package for the Arduino 101 board fails with error:
CRC doesn't match, file is corrupted. It may be a temporary problem, please retry later.

The file causing the problem is:
https://github.com/01org/corelibs-arduino101/archive/2.0.2.zip

The checksum in package_index.json is
cc81c46b5da7d7ec94e672f1cca509b03827828c60435d75eb60e7b9ba9da253
while the actual checksum of the file is
2cea02dee6959f784c6bb6bdb2dc0eafa4c4d0ce5539da9cfa64dd6ce50939fe

When downloading the file with wget, I see a 301 redirect to https://github.com/arduino/ArduinoCore-arc32/archive/2.0.2.zip    Could this be an issue with a renamed project as in https://github.com/arduino/Arduino/issues/4475 ?

I didn't find any documentation about a manual install or bypassing the CRC check, and ended up making my own local package.json with the proper CRC in it.   Is this really the only way to work around this issue?

facchinm, 2017-12-02:
Hi @frankracis ,
thanks for reporting. I just pushed the updated crc (and size) for arc32 core 2.0.2
The problem arises from using github's automatic tarballs, which are being recreated every time a tag is pushed (so also when the repository changes its name). 
Core 2.0.2 should be the only affected but I need to double check all the other links so I'm leaving the issue open until everything is back to normal.

Lothgar, 2017-12-11:
Having the same issue when trying to install core 1.0.7 is there any workaround?

facchinm, 2017-12-12:
Should be fixed now for all packages, please reopen if you should encounter any problem in downloading any core revision

closed 2017-12-12 by facchinm