arduino/Arduino #6982: java.lang.NullPointerException when install-boards

patrickelectric, 2017-12-02:
```
arduino --install-boards esp8266:esp8266
Picked up JAVA_TOOL_OPTIONS: 
Loading configuration...
Initializing packages...
Preparing boards...
Downloading platforms index... 
Downloading platforms index... Downloaded 0kb of 197kb.
Downloading platforms index... Downloaded 10kb of 197kb.
Downloading platforms index... Downloaded 12kb of 197kb.
Downloading platforms index... Downloaded 19kb of 197kb.
Downloading platforms index... Downloaded 29kb of 197kb.
Downloading platforms index... Downloaded 40kb of 197kb.
Downloading platforms index... Downloaded 51kb of 197kb.
Downloading platforms index... Downloaded 61kb of 197kb.
Downloading platforms index... Downloaded 71kb of 197kb.
Downloading platforms index... Downloaded 80kb of 197kb.
Downloading platforms index... Downloaded 83kb of 197kb.
Downloading platforms index... Downloaded 90kb of 197kb.
Downloading platforms index... Downloaded 97kb of 197kb.
Downloading platforms index... Downloaded 99kb of 197kb.
Downloading platforms index... Downloaded 110kb of 197kb.
Downloading platforms index... Downloaded 120kb of 197kb.
Downloading platforms index... Downloaded 130kb of 197kb.
Downloading platforms index... Downloaded 140kb of 197kb.
Downloading platforms index... Downloaded 150kb of 197kb.
Downloading platforms index... Downloaded 159kb of 197kb.
Downloading platforms index... Downloaded 169kb of 197kb.
Downloading platforms index... Downloaded 176kb of 197kb.
Downloading platforms index... Downloaded 180kb of 197kb.
Downloading platforms index... Downloaded 184kb of 197kb.
Downloading platforms index... Downloaded 194kb of 197kb.
Downloading platforms index... Downloaded 197kb of 197kb.
Downloading platforms index... 
Downloading platforms index... Downloaded 0kb of 0kb.
Downloading platforms index... Downloaded 0kb of 0kb.
java.lang.NullPointerException
	at java.util.Collections.sort(Collections.java:175)
	at processing.app.Base.<init>(Base.java:310)
	at processing.app.Base.main(Base.java:144)
make: *** [deps] Error 255
```

This running on [travis](https://travis-ci.org/IBEXcps/HaLiO-sw/builds/310596858).

per1234, 2017-12-02:
Fixed by https://github.com/IBEXcps/HaLiO-sw/pull/23

matthijskooijman, 2017-12-03:
Shouldn't the Arduino IDE give a proper error message instead of a NullPointerException? IMHO content (e.g. platform definitions) should never cause (these kinds of) exceptions... @facchinm, @cmaglie, what would you think?

facchinm, 2017-12-04:
I agree, a more meaningful error (like `platform esp8266:esp8266 is unknown`) would be nice