arduino/Arduino #6983: Arduino IDE Extremely Slow and Buggy

KevinColeman14, 2017-12-02:
I have one computer that arduino IDE does not like to work on. It is my newest computer with the fastest processors and I cant figure out the issue.

If I compile a sketch. it takes 30-40 seconds. Same to upload. 
If I open another sketch, that's another 30-40 seconds.
If I try to move the window that IDE is opened in, it will break.
When I close arduino IDE, It freezes and takes several minutes to close.

I have tried reinstalling arduino IDE with no success. 
Currently using arduino 1.8.5

PaulStoffregen, 2017-12-02:
Maybe this is Windows?  Maybe you also have malware or an aggressive anti-malware program running that's adding a huge burden to all file I/O?

KevinColeman14, 2017-12-03:
I have recently updated windows 10 to the latest version. This did not fix my issue.

I am running AVG as my anti virus software. Same as my other conputers that have no issue. And all other programs and operation on my computer run flawlessly. I'm completely clueless.

feikname, 2017-12-03:
Do you happen to have a very large folder (as in, with a lot of files) in your Arduino folder (the one that contains sketchs and libraries)? That's a possible cause.

PaulStoffregen, 2017-12-04:
Perhaps try temporarily disabling Windows Defender and AVG.  Perhaps disconnect from the net while these are disabled.  The idea is merely to see if they are causing the slowdown.

Windows Defender (which is built into Windows 10, so you **DO** have it) in particular as been reported to cause tremendous slowdown in some computers.

I can tell you it runs just fine here on Windows 10 (well, about half the speed of Linux, but that's "normal")

KevinColeman14, 2018-04-06:
I'll definitely try disabling each and see if it helps. I'll post results.

sandeepmistry, 2019-09-16:
@KevinColeman14 any updates on this?

KevinColeman14, 2019-09-17:
> @KevinColeman14 any updates on this?

Everything I tried had no effect on the speeds. I switched to a different computer for a while and either a windows update or an IDE update fixed the issue. I no longer have the issue. Windows defender is still running. The files are still exactly where I had them. Nothing else changed. It just started working. -_- I hate not know the real answer. Sorry I couldn't be of more help.

closed 2019-09-17 by KevinColeman14