arduino/Arduino #6985: [Library Manager] Please Add MorePins library.

dariusmihai, 2017-12-03:
GitLab Repository: https://gitlab.com/dariusmihai/Arduino_Lib_MorePins.git

MorePins is a library for increasing the number of pins on your Arduino - via Shift Registers - 
Only tested on Arduino Uno for up to 32 input pins. 
Theoretically the number of pins can be as high as you want. More info on https://gitlab.com/dariusmihai/Arduino_Lib_MorePins. 

Currently only input pins are supported. 
Soon to come - output pins.

per1234, 2017-12-03:
You need to create a tag that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

dariusmihai, 2017-12-03:
Thank you for the quick reply, @per1234 
The tag is created.

Sorry for missing it in the first place.

Have a great day.

cmaglie, 2017-12-04:
Done!

closed 2017-12-04 by cmaglie