arduino/Arduino #6987: FTDIPORT/VCP drivers shipped with Arduino are six years out of date

ariccio, 2017-12-05:
The `ftdiport.inf` file that the Arduino installer just dropped on my machine says that the driver version is 03/18/2011,2.08.14.

The latest on their website for Windows is 2.12.28, released 2017-08-30. This is six years old. The release notes a whole ton of fixed bugs (including fixed BSODs!), and Windows 10 certification. Please update.

...and it's probably a good idea to check the other drivers too!

ArsenioDev, 2018-03-22:
Thing is new driver versions have an issue where they brick any and all fake FTDI chips and that is a source of complaining on the forums

facchinm, 2019-09-20:
All the other drivers are signed by us; the FTDI one is particularly difficult to update without breaking boards / licenses. I'm closing the issue as wontfix, please reopen if the problem becomes much bigger

closed 2019-09-20 by facchinm