arduino/Arduino #6990: Digital output 13 turns on and off fast during arduino boot.

dangelobr, 2017-12-05:
Arduino Uno R3.
Digital output 13 turns on and off fast during arduino boot.
All other outputs are OK.
I tested them all with the same code, changing only the port number.
After the boot it works normally.

void setup ()
{
pinMode (13, OUTPUT);
digitalWrite (13, LOW);
}

matthijskooijman, 2017-12-05:
Pin 13 is connected to the LED. IIRC the bootloader blinks that led on startup. If this is a problem, you should probably recompile your own bootloader to not blink. Easier would be to use a different pin so this is not a problem.

I don't think there's a bug here. If you agree, please close this issue.

dangelobr, 2017-12-05:
That makes perfect sense. I looked at the layout of the board more closely and you're right. Thank you for your rating I'll close the question.

closed 2017-12-05 by dangelobr

per1234, 2021-09-07:
Related; https://github.com/arduino/Arduino/issues/6990