arduino/Arduino #6992: Request to add KniwwelinoLib to Library Manager

lazyzero, 2017-12-07:
Hello,
can you please add the library for for our Kniwwelino Board to the Library Manager?
URL to access the lib: https://github.com/LIST-LUXEMBOURG/KniwwelinoLib
more information on the Kniwwelino Board:
http://www.kniwwelino.lu

Kind regards 
Christian

per1234, 2017-12-07:
You need to create a release or tag that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

lazyzero, 2017-12-07:
Hi, we made a release now.

haarism19, 2017-12-08:
lazyzero can you help me out . I'm a newbie so I don't much, what does "create a release " mean ??? Do I have to include a license.txt file in my repository or just change the version in librabry.properties file. Have a look if possible at my library if you have time : 
https://github.com/haarism19/test_for_slab

lazyzero, 2017-12-08:
Hi @haarism19 you can make a release for your repository on https://github.com/haarism19/test_for_slab/releases

cmaglie, 2017-12-14:
Done!

closed 2017-12-14 by cmaglie