arduino/Arduino #6994: Request to add Slab to Library Manager

haarism19, 2017-12-08:
Hello,
can you please add the library for our Slab Board to the Library Manager?
URL to access the lib: https://github.com/haarism19/test_for_slab
more information on the Slab Board:
http://startoonlabs.com/Innovation/

Kind regards
Haaris

per1234, 2017-12-08:
You need to create a release or tag that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

haarism19, 2017-12-08:
Done, also I wanted to delete this whole repository after its being tested and create a proper library for Slab-32 board. Can you guys delete it from the server ?

haarislabs, 2017-12-08:
Done.

haarism19, 2017-12-08:
Done

per1234, 2017-12-08:
>Done, also I wanted to delete this whole repository after its being tested and create a proper library for Slab-32 board. Can you guys delete it from the server ?

Let's just wait until you have your finished library. It's just a big waste of time for us to add a do-nothing library and then have to delete it. I suspect you are actually wanting to implement Boards Manager installation for your 3rd party board and that has nothing to do with Library Manager and will be handled without any need for action from the Arduino team. As I recommended in your other issue (https://github.com/arduino/Arduino/issues/6997), please start a thread on the Arduino Forum and I'll help you out over there.

closed 2017-12-08 by per1234

haarislabs, 2017-12-11:
Thanks a lot sorry for being a newbie. I'm a student so I'm new to this so I sound a bit ignorant but I will make sure that next time I don't waste your time