arduino/Arduino #6995: arduino store

aigore, 2017-12-08:
Hello, I'm writing here just because the store section of the official website of arduino seems broken.
I'm trying to trace an order but, as soon as I try to go to the section "view your orders" every browser I use gives me back "too many redirects".
I tried also erasing cookies and history.
Also trying to write to the store support gives back that error
Maybe from here I can reach someone that can figure out how to fix the store section.
Thanks
My best

aigore, 2017-12-08:
Hi, I jest reopened the issue because I closed it by accident.
Anyway the store today is working and I was able to track an order.
I leave it open so who should have a look to it is been aware of potential issues on the website.

per1234, 2021-04-06:
Closing because there was a cluster of reports around the same few days, then nothing since, which leads me to believe the problem was resolved.

closed 2021-04-06 by per1234