arduino/Arduino #6998: Arduino Software and Windows cannot connect to my Arduino UNO

playzocker22, 2017-12-08:
Sup guys,

yesterday I tried to install Arduino IDE 1.8.5 on Windows 10 Pro (1709, Build 16299.98) and the installation went down perfectly fine, I also installed all the requested drivers. But as soon as I plugged in the Arduino Windows refused to do s**t with the Arduino and gave me the error message (Translated from German): "Error while trying to get the USB-Device Descriptor, Code 43". 

I know that this is a driver problem and I tried to solve it by disabling the automatic driver install function provided by windows but it still didn't work. Then I tried to locate the drivers manually (In the device manager) but Windows refused to load the Arduino drivers and gave me this message: The best drivers for this device are already installed. 
I remember that in an earlier version of Windows there was an option to force Windows to load drivers manually but this option is gone now, so is there another way to get the drivers working? I really dont know how to at this point.

With best regards
playzocker22

facchinm, 2017-12-11:
Hi @playzocker22 ,
the drivers for all original Arduino board are already contained in Windows 10 so there is no need to install extra drivers. Could you try plugging the board on another PC and check if it works? Also try changing the USB cable because it is a know source of problems. If this doesn't work, please contact support@arduino.cc for further assistance.

closed 2017-12-11 by facchinm

Karan231299, 2021-07-11:
@playzocler22 @facchinm 
Even i have windows 10 pro and i am using arduino 1.8.5. I am unable to select the port and hence i tried updating the driver from device manager. i am receiving the same message. i have no clue what to do. I have spent almost 7 hours on this.