arduino/Arduino #7002: Request to add library Beacon to the library manager

bengtmartensson, 2017-12-09:
This is a request to add the library [Beacon](https://github.com/bengtmartensson/ABeacon)  to the library manager. This is an AMX compatible beacon for Ethernet "things", supporting the libraries Ethernet and Ethernet2.

per1234, 2017-12-09:
I could be wrong, but I suspect there will be a problem with your [`Version-1.0.0`](https://github.com/bengtmartensson/ABeacon/releases/tag/Version-1.0.0) tag because that's not a semver compliant version, as [required by Library Manager](https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ). I know that Library Manager can handle the v1.0.0 style of version notation so I could be wrong about this. I'm sure cmaglie can provide a definitive answer when he comes through to add it to the index.

bengtmartensson, 2017-12-10:
The version is defiined  in `library.properties` and is "1.0.0". However "Version-1.0.0" is the name of the Github tag. I use the same scheme in my other submitted libraries, Infrared and DecodeIR.

cmaglie, 2017-12-14:
Done!

closed 2017-12-14 by cmaglie