arduino/Arduino #7003: Please add BlueDot_BME680 to Library Manager

BlueDot-Arduino, 2017-12-09:
Could you please add the BlueDot_BME680 library to the Library Manager?
https://github.com/BlueDot-Arduino/BlueDot_BME680

Thanks!

cmaglie, 2017-12-14:
Done!

closed 2017-12-14 by cmaglie

njh, 2020-07-23:
It looks like this repo has now been deleted.
Please could you provide some feedback on why it was deleted?

BlueDot-Arduino, 2020-07-23:
Hello Nicholas,
 
the codes from this library were not working properly and I decided not to develop it further. Could you please remove it from the Library Manager?
 
Thank you!
 
 
Thiago Barros
BlueDot UG (haftungsbeschränkt)Obernzeller Str. 281549 MunichGermanyEmail:thiago.barros@bluedot.spaceWeb:[ www.bluedot.space ]( http://www.bluedot.space/ )

njh, 2020-07-23:
I think you might need to create a new issue.

There is some information here:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ#sorry-i-did-something-wrong-how-can-i-change-or-unpublish-an-already-published-library