arduino/Arduino #7007: Arduino Editor

rosiedaisy, 2017-12-10:
I have a sketch that works on my "Arduino IDE 1.8.5" but when i copy this sketch onto the "Arduino Web Editor" it gives me this error....... "Multiple libraries were found for "LiquidCrystal_I2C.h"
Used: /home/admin/builder/opt/libraries/latest/jm_liquidcrystal_i2c-1-0-0
Not used: /home/admin/builder/opt/libraries/latest/liquidcrystal-i2c-1-1-2"  The sketch runs with no problem when loaded into my Nano from the IDE 1.8.2 so I know it's not the sketch.  How do i update the libraries or remove one?

per1234, 2017-12-10:
That's not an error, it's just a helpful message. As long as it's picking the library you intended then you can just ignore it.

closed 2017-12-10 by per1234

rosiedaisy, 2017-12-10:
It won't load the sketch into my board. Thanks for the info. Going back to using I DE on pc.-------- Original message --------From: per1234 <notifications@github.com> Date: 12/10/2017  12:19 PM  (GMT-05:00) To: arduino/Arduino <Arduino@noreply.github.com> Cc: rosiedaisy <rosiedaisy@earthlink.net>, Author <author@noreply.github.com> Subject: Re: [arduino/Arduino] Arduino Editor (#7007) 
That's not an error, it's just a helpful message. As long as it's picking the library you intended then you can just ignore it.

per1234, 2017-12-10:
Won't load indicates there is indeed an error but the message you posted does not directly tell us what it is (though it is likely a clue to the cause). I'm 99% I know the problem and a solution and am happy to help you out. However, since that will fall more under the heading of support and this issue tracker is only for bugs and feature requests, it's better that be done on the Arduino forum. So if you want help getting your sketch working with the Arduino Web Editor, please post on the Arduino forum in the [Create > Editor section](http://forum.arduino.cc/index.php?board=101.0) and we'll continue the discussion over there.

And if I'm correct in my guess, the Arduino Web Editor developers are already aware of the issue since it has been [discussed ](http://forum.arduino.cc/index.php?topic=450612.msg3468547#msg3468547)with them previously on the Arduino Forum.