arduino/Arduino #7008: Adding NewPing Library to the Library Manager

teckel12, 2017-12-10:
@per1234 I'm trying to get this working and pushing the source and you locked the issue?  Very confused why this was done.  Just trying to sort this out....

per1234, 2017-12-10:
My read of https://github.com/arduino/Arduino/issues/5798 was that you had no interest in making the changes to your repository necessary for inclusion in the Arduino Library Manager index and thus the discussion was dragging on for no productive reason, resulting in spam emails being sent to this repository's 891 subscribers.  Since it sounds like this may not be the case anymore, I'll unlock that issue again. Once your repository meets all the requirements listed in the Library Manager FAQ:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ
you can comment there and I'll reopen it.

closed 2017-12-10 by per1234

teckel12, 2017-12-10:
@per1234 Dragging on?  I just learned about this an hour ago.  I believe I'm being misunderstood.  I have the source for my library now on BitBucket and for the last hour I've been trying to figure this out and resolve it.

PaulStoffregen, 2017-12-10:
Will Arduino accept your library into their library manager when your license terms forbid any 3rd party distribution?

teckel12, 2017-12-10:
@PaulStoffregen Good point, corrected.