arduino/Arduino #7010: [OSX] Allow usage of CommandLineParser.java without showing splashscreen

jpmckearin, 2017-12-11:
https://github.com/Microsoft/vscode-arduino/issues/453

It seems like using the command line for external actions (not in the IDE) should not open a splashscreen in OSX. This splashscreen causes the 3rd party application to lose focus which can be quite annoying when using spaces in OSX.

facchinm, 2017-12-11:
Hi @jpmckearin ,
the problem is that the splash is handled by the laucher (`launch4j`) and not by Arduino itself, so we have no control if it should be displayed or not (on OSX). As a workaround, launching `pde.jar` directly does not show the splash.
However, every PR is accepted (as per Help wanted tag on https://github.com/arduino/Arduino/issues/1970)

closed 2017-12-11 by facchinm

jpmckearin, 2017-12-11:
This is what I was looking for. I'm not sure how I didn't find this when searching. I will send this upstream. Thanks @facchinm 
https://github.com/Microsoft/vscode-arduino/pull/451

jpmckearin, 2017-12-11:
It looks like someone else already got around to this. I must have missed it when searching. Thanks for the help @facchinm