arduino/Arduino #7011: Error compile Windows 10 and esp8266

sabas1080, 2017-12-11:
Hi 

### Basic Infos

#### Hardware
Hardware:			ESP12
Core Version:      	2.3.0

### Description

Problem description

### Settings in IDE

Module:  			Wemos D1
Flash Size: 		4MB
CPU Frequency:		80Mhz
Flash Mode:			qio
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		nodemcu

### Sketch
```
Example Blink
```

### Debug Messages

```
unable rename

arduino.ar reason: File exists
```

I already tried running as administrator, in Arduino IDE 1.8.4 and 1.8.5 installer and portable #2987

[image]

Also try cleaning and deleting Arduino15 and the same error

Thanks

facchinm, 2017-12-12:
Hi @sabas1080 ,
the line showing the problem is https://github.com/esp8266/Arduino/blob/master/platform.txt#L84 . However, it only happens on the third execution of that recipe, which makes me think about some antivirus activity starting on that event (rapid modification of the same file in TEMP folder).
Can you try rerunning the compilation with your antivirus off? Thanks

sabas1080, 2017-12-13:
Remove antivirus is solution, thanks!

closed 2017-12-13 by sabas1080