arduino/Arduino #7016: Power design issue/bug using the regulator

newpie18, 2017-12-12:
HI Arduino, 

I am using the Arduino Micro for a project and I noticed the following concern , please correct me if i am wrong, there is two options to power on the board 

1- External Power supply. The Board uses the LP2985- 33DBVR voltage regulator to regulate the Vin to +5 V
2- It uses the USB +5 V input to power on the board instead of the power supply.

Now the problem I noticed happens when the USB is used to power on the board. The VUSB is connected to the +5V which is connected to the Regulator output. That means when using the USB as a power source the Regulator output is 5v and the regulator input is 0V.  Is that going to cause any issue in the future for long term use?

After reviewing the Regulator datasheet it says that if the Vout is higher than the Vin this cause a diode forward biased and cause a parasitic silicon-controlled rectifier to latch , resulting in a high current  to flow from the output to the input , and it is recommended to connect a Schottky diode between the regulator Vout and Vin  to prevent  damage the regulator. 

I don't see any thing mentioned about the Schottky diode on the ardunio micro datasheet , pin diagram or circuit diagram. 

Can you please let me know if they already fixed this problem or no ?

if no, Who should I contact , If I would like to order few hindered boards with the modified design -adding the schottky  ?

per1234, 2017-12-13:
Also posted at: http://forum.arduino.cc/index.php?topic=516742

newpie18, 2018-01-24:
this is issue is resolved , the company contacted me and I Miss understand the whole issue, there is no issue in the design .

Thanks Arduino

closed 2018-01-24 by per1234