arduino/Arduino #7017: Arduino pins go high on boot and stay there

SkyFoxGR, 2017-12-12:
Arduino pins go high on boot and stay there
I have set the pin on start up to be low but stile they are high.
Some times after i press more than 5 times the reset button it works but now it stop complete to work even 100 times.
2 diffrent boards and 1 is new

int tim = 500 ; //einai to diastima pou mesolavei gia na anapci to ena fos meta to allo 500ms0
int tim1 = 1000 ; //einai to diastima pou mesolavei gia na kinithi to ena svisto fos  500ms
int tim2 = 1000 ; //einai to rithmos svisimatos apo to kaintro  500ms
int timm = 6; //10 defterolepta * 6= 60 deuterolepta  //einai i diarkia pou paramenoun anamena xoris na perasi amaxi se ms
int    tim3 = 15000 ;

int con1 = 16; //10 defterolepta * 6= 60 deuterolepta  //einai i diarkia pou paramenoun anamena xoris na perasi amaxi se ms
int    conn1 = 15000 ;

int con = 0; 
int    conn = 0 ;

int timm1 = 0;
long tim4 = 240000 ;
 int tt = 10;  //
 int ttt = 11; // 
int tt1 = 13;  //
int ttt1 = 12; //
int AAA = A1; //
int led1 =  A0;      
int led2 =  2;
int led3 =  3;
int led4 =  4;
int led5 =  5;
int led6 =  6;
int led7 =  7;
int led8 =  8;
int led9 =  9;
int buttonState = LOW;
int buttonState1 = LOW  ;
boolean mm = false;
int coun = 0 ; 

boolean sta = false;
boolean sta1 = false;
boolean sta2 = false;
boolean sta3 = false;
boolean sta4 = false;
boolean sta5 = false;
boolean sta6 = false;
boolean sta7 = false;

int ll =0; 
int lll =0;
int ll1 = 0;
int lll1 = 0;
void setup() {
pinMode(tt, OUTPUT);
pinMode(ttt, OUTPUT);
pinMode(tt1, OUTPUT);
pinMode(ttt1, OUTPUT);
  pinMode(led1, OUTPUT);
pinMode(led2, OUTPUT);
pinMode(led3, OUTPUT);
pinMode(led4, OUTPUT);
pinMode(led5, OUTPUT);
pinMode(led6, OUTPUT);
pinMode(led7, OUTPUT);
pinMode(led8, OUTPUT);
pinMode(led9, OUTPUT);
  pinMode(AAA, INPUT);
digitalWrite(led1, LOW);
digitalWrite(led2, LOW);
digitalWrite(led3, LOW);
digitalWrite(led4, LOW);
digitalWrite(led5, LOW);
digitalWrite(led6, LOW);
digitalWrite(led7, LOW);
digitalWrite(led8, LOW);
digitalWrite(led9, LOW);
}

liamkinne, 2017-12-13:
Few questions:

- Which pins exactly are staying HIGH?
- How are you measuring if a pin is HIGH or LOW?
- Do they go LOW after about a second?

SkyFoxGR, 2017-12-17:
led1 - led9
led1 - led9 give power to soild state relays and then to lamps. those lamps stay high forever.
if i start the program then is working fine but if the program go in standby mode and become idle then after 7 sec the lamps go from low to high themselves.
Sometimes after I press more than 5 times the reset button it works but now it stop complete to work even 100 times.
for test i have put on extreamly simple program and it have work perfectly.
is like the booting system have gone complete rogue(void setup).

liamkinne, 2017-12-18:
Sounds like it is to do with the circuit that you have and not the code.

Here is a page that details why your solid state relays aren't turning off:
http://www.omron.com.au/service_support/FAQ/FAQ02155/index.asp

If that page doesn't give you answers, attach an image or diagram of your circuit and I might be able to spot any problems.

closed 2017-12-18 by per1234