arduino/Arduino #7018: Fried Arduino mega 2560 / Ramps 1.4

Gorgoton, 2017-12-13:
First of all, hi and thank you to help me, 

Actually, I don't know if my question is ask in the right project, if not please refer me to the right place.

**Context:** I know how to use an Arduino with his ramps 1.4 for a 3D printer but when we talk about repairing and understand how it works, I am real newby. 
So, I have been printing pieces for maybe two weeks and while the printer was printing, I touch the board to see if the fans were cooling the board correctly. (I have already printed maybe 3x 4 hours + print WITH the fans) The printer suddenly stop printing and the LCD screen became all white. As far as I know, any cable or wire have been disconnected during the manipulation. There was a fried electronic / plastic smell coming out of the power supply side of the board. At first sight, there is nothing different on the board appearance (Arduino mega and ramps 1.4).

I have no idea which component could have fired. Please give me a starting point about the problem and how could I repair it. 

Thanks a lot

per1234, 2017-12-13:
This issue tracker is only to be used to report bugs or feature requests. This topic is more appropriate for the Arduino Forum. I'm sure we'll be able to help you with your problem over there.

Please do this:
1. Read http://forum.arduino.cc/index.php?topic=148850
2. If you haven't already done so, create a new thread in the appropriate section of the forum (http://forum.arduino.cc/), following all the rules. Be sure to post your code, using code tags (`</>` button on the toolbar).

closed 2017-12-13 by per1234

Gorgoton, 2017-12-13:
Ok thank you