arduino/Arduino #7024: 1.8.5: impossible to install/update libraries

shinyless, 2017-12-14:
I've been uninstalling and reinstalling Arduino completely, removed the Arduino folder in LocalData, removed my sketch folder, nothing helps: I can't install or update arduino libraries at all on Windows 10.
Even tried to rollback to version 1.6.9, still not luck. It was working fine until I installed 1.8.5

Each time I try to install/update a lib, it creates an empty folder with a random number (i.e, "arduino_708697") in the libraries folder, then shows this error dozens of times:

Invalid library found in C:\Users\me\Documents\Arduino\libraries\arduino_708697: C:\Users\me\Documents\Arduino\libraries\arduino_708697

If I delete the folder and try again, it keeps recreating empty folders with "arduino_" and 6 random numbers.

facchinm, 2017-12-14:
Hi @shinyless ,
which library are you trying to install? The folder `arduino_randomnumbers` is generated when you already have a folder with the library name inside your `$sketchbook/libraries` folder. Anyway, it should behave exactly as a normal library, so probably the one you are trying to install is somehow malformed.

shinyless, 2017-12-14:
Hi,

It happens with any library I try to install/update: Bridge, Firmata,
LiquidCrystal, Robot Control, Robot IR Remote, Robot Motor, SD, Synapse,
and so on.

And fwiw: my $sketchbook/libraries folder is completely empty.

facchinm, 2017-12-18:
Just to check, did you try installing a library after pausing your Antivirus? The fact that folders are empty could be related with some sort of "aggressive quarantine" that an external program is trying to apply. Unfortunately you are the only user reporting this kind of problem so I have no idea of what is possibly going wrong.

shinyless, 2017-12-18:
Ah, very good point ! And indeed that was it. My antivirus (bit defender)
was preventing access to the folder. Tweaked the security/file access
settings and now it works as intended.

closed 2017-12-18 by facchinm